arduino-libraries/WiFiNINA #5: Static IP address assignment not working

luigisaetta, 2018-09-01:
Bard used: MKR WIFI 1010

According to WIFI.config() documentation it should be possible to connect to WIFI assigning a fixed IP address (no DHCP) calling WIFI.config method *before* calling WIFI.begin

actually, I have tried and it doesn't work and always get from the router (DHCP) a different IP address from the one I specified.

void setup() 
{
 Serial.begin(115200);

 WiFi.config(ip, gateway, subnet);
 
 WiFi.begin(ssid, pass);

 // connect to MQTT broker...
 mqttClient.begin(MQTT_BROKER_IP, net);

 // register the callback
 mqttClient.onMessage(messageReceived);

 connect();

 pinMode(pinControl, OUTPUT);
 pinMode(pinInternal, OUTPUT);

 digitalWrite(13, LOW);
}

sandeepmistry, 2018-09-21:
We have some firmware changes to fix this, however are still waiting on an updater tool update (https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/pull/21) to roll them out.

luigisaetta, 2018-09-22:
> We have some firmware changes to fix this, however are still waiting on an updater tool update (arduino-libraries/WiFi101-FirmwareUpdater-Plugin#21 <https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/pull/21>) to roll them out.

Ok, thanks

Pesasse let me know when we can test it… we have an importante demo with manu Arduino Wifi 1010 around and it is a little bit difficult
To manage everything with IP floating.

We can survive, but it would be nice to be able to fit IP.

Thanks and best regards

Luigi

facchinm, 2018-10-08:
Hi @luigisaetta ,
I'm setting up the documentation for the updater. In the meantime, please download the hourly version of the IDE which already contains the FW Updater and update WiFiNINA library to 1.2.0 (via library manager).
You'll get two additional sketches, flash the one called FirmwareUpdate and then run the updater from Tools menu / WiFi101 / WiFiNINA Firmware Update.

sandeepmistry, 2018-10-09:
There is a post on the forum with more info: https://forum.arduino.cc/index.php?topic=572378.0

luigisaetta, 2018-10-09:
ok thanks, I'll try right now

luigisaetta, 2018-10-09:
Doesn't seems to work !!!

First: the documentation is not enough clear. As far as I understand the FirmwareUpdater sketch is the one coming form library Wifi101. Unless I have done something wrong. (no rack of FirmwareUpdater in WifiNIna.

Second: After I have uploaded the above mentioned sketch (from WiFI1010, I'm able to test OK the connection, but then, when I try the firmware update I get: "Error while erasing flash memory". Fortunately, I.m able to load back the original sketch and it works

don, 2018-10-10:
@luigisaetta This worked OK for me once I got the Hourly build of the Arduino IDE. After updating the IDE the Tools menu says "WiFi101 / WiFiNINA Firmware Updater". Load the FirmwareUpdater sketch, then flash the new firmware. 

http://forum.arduino.cc/index.php?topic=572378.0

luigisaetta, 2018-10-10:
@don I have read your post. Don't know. I have tried 3 times, but every time the same result. 
I'm able to launch the new Firmware Updater, but it says: Error while erasing flash memory

facchinm, 2018-10-10:
@luigisaetta FirmwareUpdater is from WiFiNINA library, version 1.2.0 (should be available for update in Library manager). The example will appear in Examples/WiFiNINA/Tools/FirmwareUpdater. It's VERY different from the WIFi101 sketchsince it must deal with a different chip.

luigisaetta, 2018-10-10:
@facchinm good suggestion... I'll try now. I have a call then, but for lunch should see something more

luigisaetta, 2018-10-10:
@facchinm Yes, I should have thought. I upgraded WifiNIna to 1.2 BEFORE your docs. Therefore even installing the hourly build of Arduino IODE, WifiNIna was not upgraded and tools directory not created.
Now I have the right skectch. I'll try firmware upgrade asap. Great, thnks

luigisaetta, 2018-10-10:
OK; worked, thnks @facchinm .
Now I'm able to setup fixed IP.

ONly side effect: WiFI.getTime() doesin't work, but I think it's due to the fact that I need to setup not only IP but also a DNS. I'll investigate. Great job folks.

sandeepmistry, 2018-10-23:
Closing, the v1.1.0 firmware update is available: https://forum.arduino.cc/index.php?topic=572378.0

closed 2018-10-23 by sandeepmistry