arduino-libraries/WiFiNINA #9: MKR1010 WiFi Latency with web response time

drakerex, 2018-09-05:
We are experience a 2 second latency with all of the basic web server code examples. Our testing environment included UniFi UAP LR and UAP-Pro's via WPA with standard security. We also tested a TP-Link WA801ND. The MKR1000 yield an almost immediate response for the SimpleWebServer. Ethernet products (Leonardo Eth and Arduino Ethernet) contain an almost immediate response similar to the MKR1000 on wireless. Sandeepmistry requested we open a ticket.

He stated "I think byte by byte reads from sockets can be optimized inside the library to avoid this behavior."

sandeepmistry, 2018-09-06:
Hi @drakerex,

I've create pull request #10 to improve this.

Would you be comfortable downloading the changes and trying them out?
- Backup the `WiFiNINA` library from your sketchbook libraries folder
- Download the zip file for the modified library: https://github.com/arduino-libraries/WiFiNINA/archive/47a6ce0432eef3e21325a2953a79ed423eb178a2.zip
- Extract the zip file and place the contents in `<SKETCHBOOK FOLDER>/libraries/WiFiNINA`

drakerex, 2018-09-06:
Absolutely. I will try tomorrow.

-drr

[Description: Description: Description: Description: Description: Description: Description: Description: Description: NCIT]
David R. Rundberg Jr.
Sr. Network Engineer, iOS App Developer, Business Consultant, Owner
C/C++, C#, Objective-C, Java, VB.Net, SQL, ASP.Net, SpriteKit, Cocos2D/X
Office: 336.431.7000 (ext. 207) - Direct: 336.442.1635
Email: david@nc-it.com<mailto:david@nc-it.com> Support: service@nc-it.com<mailto:service@nc-it.com>

                                      NCIT is a division of Synergy Source Group, Inc.

                                                                          [cid:image002.png@01D1D6C9.7E712590] <https://www.facebook.com/pages/NCIT/288085724634086>

                P Please consider the environment before printing this e-mail.

drakerex, 2018-09-07:
This incredibly reduced the latency. It matches our tests of the MKR1000 and similar ethernet products.

-drr

[Description: Description: Description: Description: Description: Description: Description: Description: Description: NCIT]
David R. Rundberg Jr.
Sr. Network Engineer, iOS App Developer, Business Consultant, Owner
C/C++, C#, Objective-C, Java, VB.Net, SQL, ASP.Net, SpriteKit, Cocos2D/X
Office: 336.431.7000 (ext. 207) - Direct: 336.442.1635
Email: david@nc-it.com<mailto:david@nc-it.com> Support: service@nc-it.com<mailto:service@nc-it.com>

                                      NCIT is a division of Synergy Source Group, Inc.

                                                                          [cid:image002.png@01D1D6C9.7E712590] <https://www.facebook.com/pages/NCIT/288085724634086>

                P Please consider the environment before printing this e-mail.

drakerex, 2018-09-07:
This incredibly reduced the latency. It matches our tests of the MKR1000 and similar ethernet products. -drr

sandeepmistry, 2018-09-11:
Closed via #10.

@drakerex thanks for testing and your feedback. Later on today they'll be a new version of the library available in the library manager with the changes to address this.

closed 2018-09-11 by sandeepmistry