arduino-libraries/WiFiNINA #12: Version 1.1.1 does not accept Scheduler

tatulea, 2018-09-13:
I have a code with a Scheduler and it does not work with 1.1.1. With 1.1.0 everything was fine.
The code that I tested is

`#include "secrets.h"
#include <SPI.h>
#include <WiFiNINA.h>
#include <Scheduler.h>

// WIFI
WiFiClient client;
IPAddress server(SERVER_ADDRESS);

void setup() {
  Serial.begin(115200);
  while (!Serial);

  //  Serial.println(sizeof(DataRecord));

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    while (1);
  }

  Serial.print("Firmware: ");
  Serial.println(WiFi.firmwareVersion());
  if ((String)WiFi.firmwareVersion() != "1.0.0") {
    Serial.println("Please upgrade the firmware");
  }

  connectToWiFi();

  Scheduler.startLoop(sendDataLoop);
}

void loop() {
  Serial.println(millis());
  delay(1000);
}

void sendDataLoop() {
  connectToWiFi();

  httpPOSTData();
  delay(1000);
}

char request_response[1024] = { 0 };
int request_response_len = 0;
int request_status = 0;
char c;

bool httpPOSTData() {
  Serial.println("Trying to send data");

  if (client.connect(server, SERVER_PORT)) {
    Serial.println("connecting...");

    client.println("POST "SERVER_PATH" HTTP/1.1");
    client.println("Host: 127.0.0.1");
    client.println("User-Agent: ArduinoWiFi/1.1");
    client.println("Connection: close");
    client.println("Content-Type: application/octet-stream");
    client.println("Content-Length: 5");
    client.println();
    client.println("abcde");
    client.println();

    request_response_len = 0;
    request_status = 0;

    while (!client.available()) {
    }

    // Getting status
    while (client.available()) {
      c = client.read();
      request_response[request_response_len] = c;
      request_response_len++;

      if (request_response_len == 12) {
        request_status = ((request_response[9] - '0') * 10 + (request_response[10] - '0')) * 10 + (request_response[11] - '0');
        Serial.print("Request status: ");
        Serial.println(request_status);

        if (request_status >= 400) {
          Serial.println("SERVER/CLIENT ERROR");
          client.flush();
          client.stop();
          return false;
        }
        else {
          memset(request_response, 0, sizeof(request_response));
          request_response_len = 0;
          break;
        }
      }
    }

    // Getting headers
    while (client.available()) {
      c = client.read();
      request_response[request_response_len] = c;
      request_response_len++;

      if (request_response[request_response_len - 4] == '\r' &&
          request_response[request_response_len - 3] == '\n' &&
          request_response[request_response_len - 2] == '\r' &&
          request_response[request_response_len - 1] == '\n') {
        memset(request_response, 0, sizeof(request_response));
        request_response_len = 0;
        break;
      }
    }

    // Geting payload
    while (client.available()) {
      c = client.read();
      request_response[request_response_len] = c;
      request_response_len++;
    }
    Serial.print("PAYLOAD: ");
    Serial.println(request_response);

    client.flush();
    client.stop();
    return true;
  } else {
    Serial.println("connection failed");
    client.stop();
    return false;
  }
}

void connectToWiFi() {
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(SECRET_SSID);

    WiFi.begin(SECRET_SSID, SECRET_PASS);
  }
  Serial.println();
  Serial.print("CONNECTED TO: ");
  Serial.println(SECRET_SSID);
  Serial.print("signal strength (RSSI):");
  Serial.println(WiFi.RSSI());
  Serial.println();
}
`

It freezes after     Serial.println("connecting...");

Rocketct, 2018-09-17:
Hi @tatulea i have tested your sketch and confirm that not works with the version 1.1.1 of the WifiNINA library, the difference between the two version is the introduction of a socket buffer in the v 1.1.1, i'm investigating in order to find a solution, thank you for the sketch.

Rocketct, 2018-09-18:
Hi @tatulea, we have identified the problem, it is a compatible problem between our new library (v1.1.1) and Scheduler library:

- The Scheduler library use malloc  for allocate the thread's stack;
- The  WiFiNINA library, in the the available function use malloc for address the buffer's memory space where are stored the received data.

Due to the malloc called by the Scheduler, all the malloc called by the wifiNINA library goes wrong and don't allocate nothing, consequently the sketch freeze in the Wifi available function.
As solution i suggest you to use the Scheduler library of Mikael Patel, that you can find here: https://github.com/mikaelpatel/Arduino-Scheduler

you have to change the startloop function `Scheduler.startLoop(sendDataLoop);` as follow:

  `Scheduler.start(setup1, sendDataLoop);`

where `setup1` is a function that work like setup of the Arduino's examples, but it also could be left void, as follow:
```
void setup1()
{
}
```

tatulea, 2018-09-18:
It's ok now, thanks!

Rocketct, 2018-09-19:
Welcome, i'll close this issue.

closed 2018-10-13 by cmaglie