arduino-libraries/WiFiNINA #13: MKR WIFI 1010, Static ip no work

CERVERO1979, 2018-09-19:
Hi, I'm using mkr1010 wicth this example.

https://www.arduino.cc/en/Reference/WiFiNINAConfig

`
#include <SPI.h>
#include <WiFiNINA.h>

// the IP address for the shield:
IPAddress ip(192, 168, 0, 177);   

char ssid[] = "yourNetwork";    // your network SSID (name)
char pass[] = "secretPassword"; // your network password (use for WPA, or use as key for WEP)

int status = WL_IDLE_STATUS;

void setup()
{ 
  // Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  WiFi.config(ip);

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:   
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  // print your WiFi's IP address:
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop () {}
`

The code connect to the router, but the ip is not the correct.

I have this.

IPAddress ip(192, 168, 1, 177);   

But , device have 192.168.1.44

sandeepmistry, 2018-09-19:
Hi @CERVERO1979,

That's for reporting this. However, https://github.com/arduino-libraries/WiFiNINA/issues/5 is already open to track this issue. Please follow it for updates.

In the future, please also review any open issues before creating a new one.

closed 2018-09-19 by sandeepmistry

CERVERO1979, 2018-09-20:
Ok... sorry .

The library is wonderfull... thanks for this.