arduino-libraries/WiFiNINA #16: failure to connect to hotspots

ubidefeo, 2018-09-21:
I have not been able to connect to WPA2 mobile hotspots.
I have a Huawei E5337 setup as 2.4GHz only and an iPhone 8 with enabled Personal Hotspot.
Any other device connects to it and so do 3 MKR1000 boards on the latest firmware 19.5.4.

Any help is greatly appreciated.

thanks
ubi

sandeepmistry, 2018-09-21:
I cna reproduce this using an iPhone 8 and iOS 12. Say tuned for more details ...

sandeepmistry, 2018-10-03:
@ubidefeo has tested a new firmware change for this, which will be part of the next release.

We'll close this when the update is available.

sandeepmistry, 2018-10-23:
Closing, the v1.1.0 firmware update is available: https://forum.arduino.cc/index.php?topic=572378.0

closed 2018-10-23 by sandeepmistry