arduino-libraries/WiFiNINA #19: TLS fails for io.adafruit.com

don, 2018-09-27:
I'm having trouble connecting with TLS on the Arduino MKR 1010. I have a sketch that sends data to io.adafruit.com via MQTT. The code works fine for a non-encrypted connection on port 1883, but fails to connect to the MQTT broker using WiFiSSLClient on port 8883. I don't think this is a server problem. I have a MKR GSM 1400 using MQTTS on io.adafruit.com.

The MQTT server appears to be using the same SSL certificate as https://io.adafruit.com so I modified the [WiFiSSLClient example](https://github.com/arduino-libraries/WiFiNINA/tree/master/examples/WiFiSSLClient) to load https://io.adafruit.com/api/docs/. The MKR 1010 connects but fails to load any content before disconnecting.

[image]

Is this a certificate problem? Any ideas on how to debug this?

Environment:
Arduino IDE 1.8.7
macOS 10.13.6
WiFiNINA 1.1.1

    /*
	This example creates a client object that connects and transfers
	data using always SSL.

	It is compatible with the methods normally related to plain
	connections, like client.connect(host, port).

	Written by Arturo Guadalupi
	last revision November 2015

    */

	#include <SPI.h>
	#include <WiFiNINA.h>

	#include "arduino_secrets.h" 
	///////please enter your sensitive data in the Secret tab/arduino_secrets.h
	char ssid[] = SECRET_SSID;        // your network SSID (name)
	char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
	int keyIndex = 0;            // your network key Index number (needed only for WEP)

	int status = WL_IDLE_STATUS;
	char server[] = "io.adafruit.com";    

	// Initialize the Ethernet client library
	// with the IP address and port of the server
	// that you want to connect to (port 80 is default for HTTP):
	WiFiSSLClient client;

	void setup() {
	  //Initialize serial and wait for port to open:
	  Serial.begin(9600);
	  while (!Serial) {
	    ; // wait for serial port to connect. Needed for native USB port only
	  }

	  // check for the WiFi module:
	  if (WiFi.status() == WL_NO_MODULE) {
	    Serial.println("Communication with WiFi module failed!");
	    // don't continue
	    while (true);
	  }

	  String fv = WiFi.firmwareVersion();
	  if (fv != "1.0.0") {
	    Serial.println("Please upgrade the firmware");
	  }

	  // attempt to connect to WiFi network:
	  while (status != WL_CONNECTED) {
	    Serial.print("Attempting to connect to SSID: ");
	    Serial.println(ssid);
	    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
	    status = WiFi.begin(ssid, pass);

	    // wait 10 seconds for connection:
	    delay(10000);
	  }
	  Serial.println("Connected to wifi");
	  printWiFiStatus();

	  Serial.println("\nStarting connection to server...");
	  // if you get a connection, report back via serial:
	  if (client.connect(server, 443)) {
	    Serial.println("connected to server");
	    // Make a HTTP request:
	    client.println("GET /api/docs/ HTTP/1.1");
	    client.println("Host: io.adafruit.com");
	    client.println("Connection: close");
	    client.println();
	  }
	}

	void loop() {
	  // if there are incoming bytes available
	  // from the server, read them and print them:
	  while (client.available()) {
	    char c = client.read();
	    Serial.write(c);
	  }

	  // if the server's disconnected, stop the client:
	  if (!client.connected()) {
	    Serial.println();
	    Serial.println("disconnecting from server.");
	    client.stop();

	    // do nothing forevermore:
	    while (true);
	  }
	}

	void printWiFiStatus() {
	  // print the SSID of the network you're attached to:
	  Serial.print("SSID: ");
	  Serial.println(WiFi.SSID());

	  // print your WiFi shield's IP address:
	  IPAddress ip = WiFi.localIP();
	  Serial.print("IP Address: ");
	  Serial.println(ip);

	  // print the received signal strength:
	  long rssi = WiFi.RSSI();
	  Serial.print("signal strength (RSSI):");
	  Serial.print(rssi);
	  Serial.println(" dBm");
	}

sandeepmistry, 2018-09-27:
Hi @don,

> The MKR 1010 connects but fails to load any content before disconnecting.

I think the connect is failing, because the `connected to server` line is not printed to the serial monitor.

I can confirm the root cert `DigiCert Global Root G2` needed for `io.adafruit.com` is not bundled with the firmware on the NINA of the MKR WiFi 1010 in the factory.

[image]

We are still working on a firmware update and certificate flashing tool for the MKR WiFi 1010, that will be equivalent to the MKR1000 flow and will update this thread once it is ready.

facchinm, 2018-10-09:
Hi @don ,
the new Nina FW 1.1.0 contains by default much more root certificates (including Digicert one) so `io.adafruit.com` will now work out of the box. You also can load other certificates using the same Firmware update utility (see http://forum.arduino.cc/index.php?topic=572378.0 for all the instructions)

closed 2018-10-09 by facchinm

don, 2018-10-09:
@sandeepmistry @facchinm I updated my board to the Nina firmware v 1.1.0 and it works great. Thanks for getting this fixed!