arduino-libraries/WiFiNINA #24: MKR WiFi 1010 RGB control

vshymanskyy, 2018-10-09:
Please add API to control the built-in RGB which is connected to pins 16,17,18:
https://content.arduino.cc/assets/MKRWiFi1010V2.0_sch.zip

facchinm, 2018-10-10:
Hi @vshymanskyy ,
the APIs are already there, only a bit hidden
https://github.com/arduino-libraries/WiFiNINA/blob/4f04bd073f1d568bcdd9273b2e45126c453f384d/src/utility/wifi_drv.h#L283
You have to call it like:
```
#include <utility/wifi_drv.h>
WiFiDrv::pinMode(25, OUTPUT);  //GREEN
WiFiDrv::pinMode(26, OUTPUT);  //RED
WiFiDrv::pinMode(27, OUTPUT);  //BLUE
...
WiFiDrv::digitalWrite(27, HIGH); // for full brightness
...
WiFiDrv::analogWrite(27, 128);  // for configurable brightness
```

I'm leaving the issue open to better document it

sandeepmistry, 2018-10-10:
Please note these API's may change or be removed in the future, as the firmware on the NINA may control the LED status directly.

vshymanskyy, 2018-10-10:
@facchinm @sandeepmistry Please then have an option to disable the built-in LED control, if possible.

theeraphong0, 2019-04-05:
> Hi @vshymanskyy ,
> the APIs are already there, only a bit hidden
> [WiFiNINA/src/utility/wifi_drv.h](https://github.com/arduino-libraries/WiFiNINA/blob/4f04bd073f1d568bcdd9273b2e45126c453f384d/src/utility/wifi_drv.h#L283)
> 
> Line 283 in [4f04bd0](/arduino-libraries/WiFiNINA/commit/4f04bd073f1d568bcdd9273b2e45126c453f384d)
> 
>  static void pinMode(uint8_t pin, uint8_t mode); 
> 
> You have to call it like:
> ```
> #include <utility/wifi_drv.h>
> WiFiDrv::pinMode(25, OUTPUT);  //GREEN
> WiFiDrv::pinMode(26, OUTPUT);  //RED
> WiFiDrv::pinMode(27, OUTPUT);  //BLUE
> ...
> WiFiDrv::digitalWrite(27, HIGH); // for full brightness
> ...
> WiFiDrv::analogWrite(27, 128);  // for configurable brightness
> ```
> I'm leaving the issue open to better document it

thank you. It work.

#include <WiFiNINA.h>
#include <utility/wifi_drv.h>
void setup() 
{
  WiFiDrv::pinMode(25, OUTPUT);  //GREEN
  WiFiDrv::pinMode(26, OUTPUT);  //RED
  WiFiDrv::pinMode(27, OUTPUT);  //BLUE
}
void loop() 
{
  WiFiDrv::digitalWrite(27, HIGH); // for full brightness
  delay(1000);
  WiFiDrv::digitalWrite(27, LOW); // for full brightness
  delay(1000);
}

lotelx, 2019-07-10:
Thank you so much! It worked for me.
But the correspondence of colors are this one for my board (Arduino UNO Wifi rev 2):
Pin 25 is for red, 26 for green and 27 blue.

TalibaniGrannyFanny, 2019-11-09:
this fix worked great for me UNTIL i updated my ffirmware at which point now it,s "no dice".....

italex33, 2020-01-25:
Yes Sir, it works!

thekunalsaini, 2020-02-28:
API is available

cansik, 2020-06-17:
@systembolaget This does not seem to work for NINA, I cant find this method in the source code:

https://github.com/arduino-libraries/WiFiNINA/search?q=setLEDs&unscoped_q=setLEDs

systembolaget, 2020-06-17:
I am using Adafruit's WiFiNINA fork. Maybe you can just add their method to your WiFiNINA library flavour, and you're all set.