arduino-libraries/WiFiNINA #27: Not declared in this scope error

tbaggs, 2018-10-15:
I'm trying to compile a simple app using the WifiNINA library.  I keep getting the following errors.  I checked the code and it seems correct.  Any ideas?

/Users/tbaggs/Documents/Arduino/libraries/WiFiNINA/src/utility/spi_drv.cpp: In static member function 'static void SpiDrv::begin()':
/Users/tbaggs/Documents/Arduino/libraries/WiFiNINA/src/utility/spi_drv.cpp:97:15: error: 'NINA_GPIO0' was not declared in this scope
       pinMode(NINA_GPIO0, OUTPUT);
               ^
/Users/tbaggs/Documents/Arduino/libraries/WiFiNINA/src/utility/spi_drv.cpp: In static member function 'static int SpiDrv::available()':
/Users/tbaggs/Documents/Arduino/libraries/WiFiNINA/src/utility/spi_drv.cpp:566:25: error: 'NINA_GPIO0' was not declared in this scope
     return (digitalRead(NINA_GPIO0) != LOW);

sandeepmistry, 2018-10-16:
Hi @tbaggs,

What board are you selecting when compiling. Currently this library only supports the MKR WiFi 1010, MKR Vidor 4000.

If you are using one of these boards, please go to the IDE preference, and beside "Show verbose output during", check the "compile" box. Then try to build the sketch again and provide all the output here.

Y0hy0h, 2019-12-10:
I just stumbled into the same problem. The MKR1000 only supports the [WiFi101](https://www.arduino.cc/en/Reference/WiFi101) library.