arduino-libraries/WiFiNINA #28: Cisco AP networks not found.

charlesbones, 2018-10-16:
Networks generated by  Cisco AirNet 3602, AP is using WPA2 encryption are not visible by the antenna on a MKR WIFI 1010. The network uses multiple channels: 1, 6, and 11.

The only networks visible are generated by D-Link and Netgear routers.

Rocketct, 2018-10-26:
Hi @charlesbones, thanks for open the issue, could you please give more information?have you try to connect to the network?  you have never seen the SSID in the scanning? which are the number of network showed? which kind of WPA2 are you using? TKIP or AES? which are the mac address of the AP?
Thank you for your availability

charlesbones, 2018-11-23:
Hi @Rocketct , The information reported by the customer is:

- Have you tried to connect to the network?. Yes, If the Arduino board is in just the right spot and at just the right angle it can see it and even connect for a brief second before disconnecting.

- Have you seen the SSID in the scanning? Yes, but it is not constantly showing in the list. The SSID of the network is "idiot".

- Which is the number of shown networks? image attached.

[image]

- which kind of WPA2 are you using? TKIP or AES? the "idiot" SSID is WPA2 AES only….no TKIP.

- which are the mac address of the AP? Still waiting for this information

Thank you

Rocketct, 2019-01-29:
Hi @charlesbones i have an update, tomorrow i'll share a nina firmware version with the debug enabled, and the instruction for the customer to report and show the results, in order to investigate on what happen module side

Rocketct, 2019-01-30:
@charlesbones the following is the firmware with the debug enabled [NINA_W102.bin.zip](https://github.com/arduino-libraries/WiFiNINA/files/2812171/NINA_W102.bin.zip)
to use it:
- Go to the folder in "UserArduinoIDEpath"/tools/WiFi101/tool/firmwares/NINA/1.2.1 ;
- Backup the file NINA_W102.bin (to restore it when the test are done) and replace it with the one attached here;
- Run the IDE;
- Upload the sketch `Example->WiFiNINA->tools->Firmware updater`;
- Upload  the new firmware using `WiFi101/WiFiNINA firmware updater` in tools ;
- Upload the sketch `Example->ScanNetworks`;

To show the log of the NINA module is required a seconds board with the sketch SerialPasstrhough uploaded with a baudrate =115200:
- if the customer have another wifi1010 he should use the example in Example->WiFiNINA->tools called  SerialNINApassthrough.

When the boards are ready, the customer have to connect some pins of the boards as following:

| Board_1(ScanNetwork)  |    Board_2 |
|---|---|
| GND        | GND |
| 14<-TX    |    13->RX |
| 13->RX    |                  14<-TX   |

Then open the serial monitor of the two boards and in the Serialmonitor of the board_2 he should see all the network detected with the ESP32's scan API.

Let me know the results.

Rocketct, 2019-06-20:
closed due to lack of feedback @charlesbones reopen when available

closed 2019-06-20 by Rocketct