arduino-libraries/WiFiNINA #30: WiFiNINA_OTA ?

Ginodel, 2018-10-19:
Hi all,

Are there plans for a WifiNINA_OTA library like you have for the MKR1000 board ? Or.. is there somewhere a library that is allowing OTA programming ?

Thanks for your input

Ginodel, 2018-10-22:
Hi Facchinm... 
I just tried the patch ... In the Arduino IDE i can see the network port... 
I was compiling the sketch and while uploading i get ...

De schets gebruikt 45136 bytes (17%)  programma-opslagruimte. Maximum is 262144 bytes.
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:

Any idea why ????

facchinm, 2018-10-23:
Which sketch are you using? Plain `WiFi101OTA` example?  Please enable verbose upload and paste here the output (should be something like `.../.arduino15/packages/arduino/tools/arduinoOTA/1.2.0/bin/arduinoOTA -address 10.130.22.193 -port 65280 -username arduino -password password -sketch /tmp/arduino_build_408639/WiFi101_OTA.ino.bin -upload /sketch -b `)

DesktopMan, 2019-06-02:
Any updates on this? "soon" is relative, but it's been a while :)

facchinm, 2019-06-17:
I know @DesktopMan , sorry for that.
The first pieces of support have already been published (megaAVR core with uart bootloader support, updated arduinoOTA utility), other are still in staging.
We'll tell everyone as soon as the merge is finished :wink:

blackketter, 2019-07-10:
Is it safe to assume that this OTA support will also apply to the Arduino NANO 33 IoT?

adams-family, 2020-02-02:
I'm also very interested in OTA updates for the Arduino Nano 33 IoT! Anyone had luck with it?

jandrassy, 2020-02-03:
for SAMD boards with NINA use WiFi101OTA library. only replace the include to WiFiNINA.h

milenovic, 2020-04-24:
Hi all,
I noticed that there is unowifirev2_ota branch which got me very excited! Is it possible to use OTA on Uno WiFi Rev2 already? I could not find any information about that branch.

adams-family, 2020-04-24:
I can confirm that OTA on the Arduino Nano 33 IoT works perfectly using this libary:

https://github.com/jandrassy/ArduinoOTA

It's enough to:
 1) Install the package using package manager
 2) Import one .h header file
 3) Add two lines of code (provided that the Arduino sketch already has a connection to the WiFi)
 4) Restart Arduino IDE and enjoy!

facchinm, 2020-04-24:
For the Uno Wifi rev2 we are rebasing that branch in these days, with the plan to release it shortly after :wink:

jandrassy, 2020-04-24:
> I can confirm that OTA on the Arduino Nano 33 IoT works perfectly using this libary:
> https://github.com/jandrassy/ArduinoOTA

it doesn't support Uno WiFi

adams-family, 2020-04-24:
@jandrassy Honestly I'm not sure what is Uno WiFi?

I got it working on a genuine Arduino Nano 33 IoT, I can flash the device even now :)

jandrassy, 2020-04-25:
> @jandrassy Honestly I'm not sure what is Uno WiFi?
> 
> I got it working on a genuine Arduino Nano 33 IoT, I can flash the device even now :)

you commented right after other user who asked about support for Uni WiFi. I thought you answered.

adams-family, 2020-04-25:
@jandrassy Sorry, no :-) I just wanted to be useful and answer my own comment from Feb 2 where I asked about the Arduino Nano 33 IoT. Sorry for the confusion.

n-gineer, 2020-07-06:
> For the Uno Wifi rev2 we are rebasing that branch in these days, with the plan to release it shortly after 😉

> At the moment a patched WiFI101OTA library can be used ([arduino-libraries/WiFi101OTA#11](https://github.com/arduino-libraries/WiFi101OTA/pull/11)) but we are going to provide some super cool NINA-specific features soon 😉

> for SAMD boards with NINA use WiFi101OTA library. only replace the include WiFi101.h to WiFiNINA.h

Just to clarify, is it possible to use the mentioned patch/editing includes to use the WiFi101OTA library on Uno Wifi Rev 2? I tried making the edits in a copy of the WiFi101OTA library, and got these errors:

`

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\...\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\...\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\...\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10813 -build-path C:\Users\...\AppData\Local\Temp\arduino_build_818150 -warnings=none -build-cache C:\Users\...\AppData\Local\Temp\arduino_cache_211078 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\...\Arduino\libraries\Arduino_WiFiOTA\examples\WiFi_Nina_OTA\WiFi_Nina_OTA.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\...\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\...\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\...\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10813 -build-path C:\Users\...\AppData\Local\Temp\arduino_build_818150 -warnings=none -build-cache C:\Users\...\AppData\Local\Temp\arduino_cache_211078 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\...\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\...\Arduino\libraries\Arduino_WiFiOTA\examples\WiFi_Nina_OTA\WiFi_Nina_OTA.ino
Using board 'uno2018' from platform in folder: C:\Users\...\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\...\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6
Detecting libraries used...
"C:\\Users\\...\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\...\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\...\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\...\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\uno2018" "C:\\Users\\...\\AppData\\Local\\Temp\\arduino_build_818150\\sketch\\WiFi_Nina_OTA.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Arduino_WiFiOTA.h: [Arduino_WiFiOTA@1.1.0]
ResolveLibrary(Arduino_WiFiOTA.h)
  -> candidates: [Arduino_WiFiOTA@1.1.0]
"C:\\Users\\...\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\...\\AppD [... cut ...]
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Using library Arduino_WiFiOTA at version 1.1.0 in folder: C:\Users\...\Arduino\libraries\Arduino_WiFiOTA 
Using library WiFiNINA at version 1.5.0 in folder: C:\Users\...\Arduino\libraries\WiFiNINA 
Using library SD at version 1.2.4 in folder: C:\Users\...\Arduino\libraries\SD 
Using library SPI at version 1.0 in folder: C:\Users\...\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\libraries\SPI 
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
`

n-gineer, 2021-02-09:
@facchinm 
If you can point me in the right direction, I'd be happy to try working on uno wifi Rev 2 support. 

I noticed the waiting for feedback label; is there anything you are still waiting for feedback on?

jandrassy, 2021-02-09:
@n-gineer did you see this https://github.com/arduino/nina-fw/issues/36 ?

jandrassy, 2021-03-10:
> > @n-gineer did you see this [arduino/nina-fw#36](https://github.com/arduino/nina-fw/issues/36) ?
> 
> @jandrassy thanks for the reply.
> 
> I've tried following the linked comments, issues, libraries, and pull requests, but I'm a little lost. It seems like [arduino/nina-fw#32](https://github.com/arduino/nina-fw/pull/32) enabled this for Uno Wifi Rev 2? If so, what library should I use for OTA?
> 
> https://github.com/arduino-libraries/WiFi101OTA? Its patch [arduino-libraries/WiFi101OTA#11](https://github.com/arduino-libraries/WiFi101OTA/pull/11)?, Just this library, WifiNINA? Your ArduinoOTA?

I meant it if you want to work on OTA support for Uno WFi

n-gineer, 2021-03-10:
@jandrassy Sorry, now I follow. This is getting outside my knowledge base, how/who can I ask more questions to get started learning what to do? Is the issue you linked referring to Wifi Rev 2 even though it only says "Uno Wifi"? Should I ask more there?

tdig2, 2021-03-24:
I recently bought the Uno WiFi Rev2 because I thought it could be programmable OTA...looks like @facchinm and @jandrassy are the only ones working on the WiFiNINA_OTA library?  This would in theory allow the Uno WiFi Rev2 to create its own Network port, correct?
Apologies if I'm not getting the vernacular correct here - new to Arduino, this is my first time trying to upload code OTA with any board.

tdig2, 2021-06-01:
Thoughts?

SpyGuy0215, 2022-04-17:
Any updates on this? It's been about a year or so, and there doesn't seem to be any way to update an Uno WiFi Rev.2 OTA as of now. Any library we can use for the time being?

Sythsaz, 2024-05-17:
Any updates on this yet?

JAndrassy, 2024-05-17:
> Any updates on this yet?

which part od it? MKR? Uno WiFi?

Sythsaz, 2024-06-06:
@JAndrassy I apologize I should have been more specific, I am referring to the Uno WiFi Rev 2 OTA support that was announced years ago with no implementations made. As like with many others, it was bought solely for that reason.