arduino-libraries/WiFiNINA #33: MKR 1010 - WiFi status always returns 1 when using reserved dhcp ip

saif76, 2018-10-21:
WiFiNiNA version : 1.2
Board : MKR 1010
Board version : 1.1

I am using a meraki acces point. I noticed that the MKR returns 1 as its wifi status whenever I turn on a DHCP reservation for it. I will test on other routers I have lying around but it consistently happens with the current WiFi setup I have. Let me know if you would like the AP logs.

saif76, 2018-10-22:
Yes I updated the firmware version to see if that would fix the problem but it did not. After that is when I reported this issue on github. I was able to set a static IP and it does work (need to test further). It fails to work when I enable dhcp ip reservation for the MKR1010 in the router.

sandeepmistry, 2018-10-23:
Do you think you can get a Wireshark capture of the DHCP messages exchanged?

saif76, 2018-10-23:
Yes I can do that. I will try to do it tonight.

saif76, 2018-10-28:
@sandeepmistry here is the packet capture https://www.dropbox.com/s/se0onu6qgekltoz/MKR1010-DHCP_fail.pcap.zip?dl=0

Rocketct, 2018-11-02:
Hi @saif76, i make some test with another brand of AP and the DHCP reservation work fine, have you tried the 1010 with another AP? could also try the reservation with other devices that are not the mkr1010? could you share the sketch that are you using?

sandeepmistry, 2019-01-30:
Closing this for now due to lack of feedback, please re-open with requested info if you are still interested.

closed 2019-01-30 by sandeepmistry

ninora, 2019-02-01:
I will share my experience hoping this helps, if not, sorry for bothering you guys. 
It was impossible to connect a MKR 1010 Wifi to Goggle Wifi access point until I reserved an address for the MKR (on the AP DHCP). From that moment and on, no issues, MKR always connects. 
I wanted to make sure the MKR will not have a problem connecting to other networks when I ship it to clients so trying to replicate the problem I unreserved the address and reset all three AP's several times, still, MKR always connected with the same IP I previously reserved (and unreserved). Finally, I reserved that address to my phone so MKR will not get it, MKR got a new IP and connected with no problems.
Bottom line, no connection until reserving an IP, afterwards, always connecting. Obviously I am concerned how it will perform when connecting to new networks.