arduino-libraries/WiFiNINA #47: Server restart

moebius911, 2019-02-07:
Hello

currently i use the function "server.begin ()" to launch the web server on my arduino and it works very well. However I noticed that after several wifi disconnection the server ended up crashing.
My problem is that I do not see any function to obtain the state of the server or to restart it.
Do you have a solution to offer me?

sandeepmistry, 2019-02-07:
Hi @moebius911,

Do you have some steps to reproduce the "I noticed that after several wifi disconnection the server ended up crashing" condition?

moebius911, 2019-02-07:
```

here is a piece of my program that I have simplified. the server crash after about 10min

#include <WiFiNINA.h>  // for use with Arduino Uno WiFi Rev2
#include <avr/wdt.h>
#include <avr/io.h>
#include <avr/interrupt.h>      // library for interrupts handling
#include <avr/sleep.h>          // library for sleep
#include <util/delay.h>

//********************************Variable globales************************************
int batterie = A0;
int hydrometrie1 = A1;
int hydrometrie2 = A2;
int hydrometrie3 = A3;
int hydrometrie4 = A4;
int hydrometrie5 = A5;
int FondCuve = 10;

// set water relays
int vanne4 = 2;
int vanne3 = 3;
int vanne2 = 4;
int vanne1 = 5;
int pompe = 6;
int chauffage = 7;
int ventilation = 8;
int relais8 = 9;

// Network
char ssid[] = "freebox_Alex";     // your network SSID (name)
char pass[] = "****************";  // your network password
WiFiClient client; // Initialize the client library
int status = WL_IDLE_STATUS;
WiFiServer server(80);
byte IP_Serveur[] = { 192, 168, 1, 252 };
int Port = 8080; 

/*****************************************************************/
/*                       Initialisation                          */
/*****************************************************************/
void setup() {
  //initialisation des dialogues
  Serial.begin(9600);  // initialize serial communication
  Serial.println(F("Initialisation en cours......."));
  pinMode(vanne1, OUTPUT);
  pinMode(vanne2, OUTPUT);
  pinMode(vanne3, OUTPUT);
  pinMode(vanne4, OUTPUT);
  pinMode(pompe, OUTPUT);
  pinMode(chauffage, OUTPUT);
  pinMode(ventilation, OUTPUT);
  pinMode(relais8, OUTPUT);
  pinMode(hydrometrie1, INPUT);
  pinMode(hydrometrie2, INPUT);
  pinMode(hydrometrie3, INPUT);
  pinMode(hydrometrie4, INPUT);
  pinMode(hydrometrie5, INPUT);
  pinMode(FondCuve, INPUT);
  digitalWrite(vanne1, HIGH);
  digitalWrite(vanne2, HIGH);
  digitalWrite(vanne3, HIGH);
  digitalWrite(vanne4, HIGH);
  digitalWrite(pompe, HIGH);
  digitalWrite(chauffage, HIGH);
  digitalWrite(ventilation, HIGH);
  digitalWrite(relais8, HIGH);  
  WiFi.lowPowerMode(); 
  timer(1); //timer interruption en sec
[... 293 lines cut ...]
/*                  Configuration de l'interruption timer                   */
/****************************************************************************/
void timer(int secs){
  while (RTC.STATUS > 0) {} // attend que tous les registre soit synchronisée
  
  RTC.PER = 1024*secs; //secs = interuption timer en secondes.
  RTC.INTCTRL = 0 << RTC_CMP_bp
  | 1 << RTC_OVF_bp; //Overflow interrupt.
  
    RTC.CTRLA = RTC_PRESCALER_DIV1_gc //NO Prescaler
  | 1 << RTC_RTCEN_bp         //active RTC
  | 1 << RTC_RUNSTDBY_bp;     //fonctionne en standby
  
  RTC.CLKSEL = RTC_CLKSEL_INT1K_gc; // 32KHz divisé par 32, fonctionne à 1.024kHz
  sei(); //active l'interruption
}
/****************************************************************************/
/*                           Interuption par Timer                          */
/****************************************************************************/
ISR(RTC_CNT_vect)
{
        RTC.INTFLAGS = RTC_OVF_bm;
        ControlClientWebPage(); //ecoute les connections clients
}
```

Rocketct, 2019-02-18:
Hi @moebius911 i had try your code and reproduced the issue, on my case happen that after the the first re connection the server was not able to response.
There is a server.status() API that return if there are  sockets opened or not, to restart on this case the only way is use again the begin, but you have to be sure about the connection, sometimes i had see the status function return 0 that means not-connected.
I have modified your sketch in order to:
- Be persistent in the module re connection, on my case some time it require more than 3 times to connect;
- Assign the same IP address to the module using the WiFi.config in ```ConnexionWifi()```;
- Wait for server begin success in ```ConnexionWifi()```, 
- Call the disconnect only one time in the loop() function.
In this way the code seems more reliable;

```c++
#include <WiFiNINA.h>  // for use with Arduino Uno WiFi Rev2
#include <avr/wdt.h>
#include <avr/io.h>
#include <avr/interrupt.h>      // library for interrupts handling
#include <avr/sleep.h>          // library for sleep
#include <util/delay.h>

//********************************Variable globales************************************
int batterie = A0;
int hydrometrie1 = A1;
int hydrometrie2 = A2;
int hydrometrie3 = A3;
int hydrometrie4 = A4;
int hydrometrie5 = A5;
int FondCuve = 10;

// set water relays
int vanne4 = 2;
int vanne3 = 3;
int vanne2 = 4;
int vanne1 = 5;
int pompe = 6;
int chauffage = 7;
int ventilation = 8;
int relais8 = 9;

// Network
char ssid[] = "ArduinoBoards";     // your network SSID (name)
char pass[] = "ArduinoccRulez";  // your network password
WiFiClient client; // Initialize the client library
int status = WL_IDLE_STATUS;
WiFiServer server(80);
byte IP_Serveur[] = { 192, 168, 1, 252 };
int Port = 8080; 

/*****************************************************************/
/*                       Initialisation                          */
/*****************************************************************/
void setup() {
  //initialisation des dialogues
  Serial.begin(9600);  // initialize serial communication
  Serial.println(F("Initialisation en cours......."));
  pinMode(vanne1, OUTPUT);
  pinMode(vanne2, OUTPUT);
  pinMode(vanne3, OUTPUT);
  pinMode(vanne4, OUTPUT);
  pinMode(pompe, OUTPUT);
  pinMode(chauffage, OUTPUT);
  pinMode(ventilation, OUTPUT);
  pinMode(relais8, OUTPUT);
  pinMode(hydrometrie1, INPUT);
  pinMode(hydrometrie2, INPUT);
  pinMode(hydrometrie3, INPUT);
  pinMode(hydrometrie4, INPUT);
  pinMode(hydrometrie5, INPUT);
  pinMode(FondCuve, INPUT);
  digitalWrite(vanne1, HIGH);
  digitalWrite(vanne2, HIGH);
  digitalWrite(vanne3, HIGH);
  digitalWrite(vanne4, HIGH);
[... 305 lines cut ...]
void timer(int secs){
  while (RTC.STATUS > 0) {} // attend que tous les registre soit synchronisée
  
  RTC.PER = 1024*secs; //secs = interuption timer en secondes.
  RTC.INTCTRL = 0 << RTC_CMP_bp
  | 1 << RTC_OVF_bp; //Overflow interrupt.
  
    RTC.CTRLA = RTC_PRESCALER_DIV1_gc //NO Prescaler
  | 1 << RTC_RTCEN_bp         //active RTC
  | 1 << RTC_RUNSTDBY_bp;     //fonctionne en standby
  
  RTC.CLKSEL = RTC_CLKSEL_INT1K_gc; // 32KHz divisé par 32, fonctionne à 1.024kHz
  sei(); //active l'interruption
}
/****************************************************************************/
/*                           Interuption par Timer                          */
/****************************************************************************/
ISR(RTC_CNT_vect)
{
        RTC.INTFLAGS = RTC_OVF_bm;
        ControlClientWebPage(); //ecoute les connections clients
}
```

Could you test and let me know if works on your side?

pozoo, 2019-02-18:
I have a similar problem: I am also using WifiNina on Arduino MKR1010 to run a simple web server. For the first couple of minutes everything works fine. The the server simply stops responding, all requests just time out. I can still ping the device.
The call:
`WiFiClient client = webServer.available(); `
simply does not seem to return any clients anymore.

Any idea how to make it respond again?

Rocketct, 2019-02-19:
Hi @pozoo could you share a simplified version of your sketch?

moebius911, 2019-02-20:
Thank you for your help, my program now seems stable
bye

closed 2019-03-08 by sandeepmistry