arduino-libraries/WiFiNINA #48: connect.client hangs on wifi reconnect for Arduino Uno Wifi Rev 2

csiro-chris-m, 2019-02-25:
This needs to run in an environment where the Wifi often drops out. However, when attempting to recover from router reboot the code stops at "client.connected()". Can you replicate?

#include <SPI.h>
#include <WiFiNINA.h>
#include <OneWire.h>
#include <Time.h>
#include <TimeLib.h>

char ssid[] = SECRET_SSID; 
char pass[] = SECRET_PASS; 
int status = WL_IDLE_STATUS;

WiFiClient client;
IPAddress ip(192, 168, 0, 33);
char server[] = "www.something.com"; 

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    ;
  }

  while (status != WL_CONNECTED) {
    status = WiFi.begin(ssid_wo, pass_wo);
    delay(5000);
  }
}

void loop() {

  while (client.available()) {
    char c = client.read();
  }
  
  client.stop();

  if (WiFi.status() != WL_CONNECTED) {  
     Serial.println(WiFi.status());
     WiFi.disconnect();
     WiFi.config(ip);
     status = WiFi.begin(ssid, pass);
     delay(5000);
  }
  else {
    delay(1000);
    // Also tried 'if (client.connect...' but using int here to see what comes back. This is where it hangs after network loss
    int cc = client.connect(server, 80);
    Serial.println("Client connect: " + String(cc));
    if (cc == 1) {
      if (client.connected()) {
        Serial.println("Connect ok");
        String celsius_string = "50";
        //Insert temp record into database
        client.println("POST /insert.php?Value=" + celsius_string + " HTTP/1.1");
        client.println("Host: www.***.***");
        //client.println("User-Agent: Arduino/1.0");
        client.println("Connection: close");
        //client.println("Content-Type: application/x-www-form-urlencoded; charset=UTF-8");
        //client.println("Content-Length: ");
        client.println();
        //Serial.println(WiFi.status());
      }
    }
    else {
      delay(1000);
      if (client.connect(server, 80)) {
        delay(5000);
        Serial.println("Attempt reconnect");
      }
    }
  }
  delay(5000);
}

csiro-chris-m, 2019-02-26:
Changing the code in loop to the following is a slight improvement in that several reconnects are dealt with fine but then it hangs again at the same statement. I am using fixed IP, also set at router.

void loop() {
//Must read any bytes still pending from client:
// "... a client is considered connected if the connection has been closed but there is still unread data"
  while (client.available()) {
    char c = client.read();
//    Serial.write(c);
  }
  
  if (WiFi.status() != WL_CONNECTED || !client.connected()) {  
     client.stop();
     client.disconnect();
     Serial.println(WiFi.status());
     Serial.println("Conn lost");
     WiFi.disconnect();
     WiFi.config(ip);
     status = WiFi.begin(ssid_rb, pass_rb);
     Serial.println("Tried restart");
     delay(5000);
     if (client.connect(server, 80)) {
       delay(1000);
     }
  }
  else {
      Serial.println("Connect ok");
  //    char c = client.read();
  //    Serial.print(c);
      //Insert something into database
      client.println("POST /some.php?Temp=" + celsius_string + " HTTP/1.1");
      client.println("Host: www.***.***");
      client.println("Connection: close");
      client.println();
      String timd = getTimeandFormat();
      Serial.println(timd + " 51");
  }
  delay(5000);
}

csiro-chris-m, 2019-02-27:
The code below which I've come up with appears to catch all dropouts. Not sure how/why but it works for me.

void loop() {
//Must read any bytes still pending from client:
// "... a client is considered connected if the connection has been closed but there is still unread data"
  while (client.available()) {
    char c = client.read();
  }
  
  if (WiFi.status() != WL_CONNECTED || !client.connected()) {  
     client.stop();
     WiFi.disconnect();
     WiFi.config(ip, dns, gateway, subnet);
     status = WiFi.begin(ssid_rb, pass_rb);
     delay(5000);
     if (WiFi.status() == WL_CONNECTED) {
       WiFi.disconnect();
       WiFi.config(ip, dns, gateway, subnet);
       status = WiFi.begin(ssid_rb, pass_rb);
       if (client.connect(server, 80)) {
         delay(3000);
       }
     }
     else
     {
       if (client.connect(server, 80)) {
         delay(3000);
       }
     }
  }
  else {
      //Insert something into database
      client.println("POST /something.php?Temp=" + celsius_string + " HTTP/1.1");
      client.println("Host: www.***.***");
      client.println("Connection: close");
      client.println();
      String timd = getTimeandFormat();
      Serial.println(timd + " 51");
  }
  delay(5000);
}

csiro-chris-m, 2019-02-27:
After more testing, this is unfortunately not yet solved. An hour or two into running the code it hangs, even though wifi is still up.

csiro-chris-m, 2019-02-27:
Four days of testing with two Arduino Uno Wifi Rev 2 have proven that WifiNina does not work reliably on that platform. If you are looking for a stable solution to log data to the Internet and that can deal with even a single network dropout then beware - this is not the way to go.

Rocketct, 2019-03-07:
@csiro-chris-m i was able to reproduce the issue it hangs on the starClient for be specific in the ```waitForSlave```, i notice that the issue happen only when the AP connection is interrupted when it try the connection to the server.

csiro-chris-m, 2019-03-07:
I have moved on to a Raspberry Pi. The Arduino is too experimental for my project.

sandeepmistry, 2019-03-21:
This issue requires a firmware change, in the next NINA firmware release a fix/workaround for it will be included.

closed 2019-03-21 by sandeepmistry

bogdosarov, 2020-04-06:
When the update will be available?

Rocketct, 2020-04-06:
it was released in version 1.2.2, now the f/w is updated to 1.3.0, available with last IDE version (1.8.12)

dionjwa, 2020-05-16:
I don't really understand:
```
if (WiFi.status() == WL_CONNECTED) {
    WiFi.disconnect();
    ...
}
```

So if Wifi is connected, you then disconnect it, before reconnecting the client? Seems scorched earth, but this stack has many random failures so 🤷

Rocketct, 2020-05-18:
hi @dionjwa can you send the line number where this piece of code is?

systembolaget, 2020-06-09:
> I don't really understand:
> 
> ```
> if (WiFi.status() == WL_CONNECTED) {
>     WiFi.disconnect();
>     ...
> }
> ```
> 
> So if Wifi is connected, you then disconnect it, before reconnecting the client? Seems scorched earth, but this stack has many random failures so 🤷

I think the disconnect before each new connection attempt is necessary, and a short delay after, as the ESP32 handles this asynchronously to what else is going on. The function below, called from loop(), works for me, when I remove the electricity from my set-up (microcontroller with additional Adafruit AirLift FeatherWing) or simulate Internet failure by disconnecting the router from the mains.

```
void connectToWLAN()
{
  // Check if the AirLift FeatherWing's WLAN module works
  if (WiFi.status() == WL_NO_MODULE)
  {
    Serial.println("Connection to AirLift FeatherWing WLAN module failed");

    // If not, indicate there is a WLAN module malfunction
    WiFi.setLEDs(0, 255, 0); // Red

    // And stop forever
    while (true);
  }

  // Return to loop() if already connected to the WLAN router
  if (WiFi.status() == WL_CONNECTED)
  {
    return;
  }

  Serial.println("Trying to connect to WLAN router");

  // Indicate there is no WLAN router connection just yet
  WiFi.setLEDs(192, 255, 0); // Yellow

  do
  {
    // Start connection to WLAN router and print a status value
    WiFi.disconnect();
    delay(1000);
    
    status = WiFi.begin(WLAN_SSID, WLAN_PASS);
    // WL_IDLE_STATUS     = 0
    // WL_NO_SSID_AVAIL   = 1
    // WL_SCAN_COMPLETED  = 2
    // WL_CONNECTED       = 3
    // WL_CONNECT_FAILED  = 4
    // WL_CONNECTION_LOST = 5
    // WL_DISCONNECTED    = 6
    Serial.println(WiFi.status());
    delay(10000);
  }
  while (status != WL_CONNECTED);

  Serial.println("Connection to WLAN router successful");

  // Indicate that the WLAN router connection was established
  WiFi.setLEDs(255, 0, 0); // Green

  // To better see the status indicating RGB LED's state
  delay(2000);
}
```

billbee, 2021-06-27:
as of 1.4.5 rev2 is still a colossal failure imho. it drops and never starts again.  I may try a mechanical timer to turn it off at night for an hour, who knows.  there is a 1.4.7 but even with the hourly build you can't get it and after all these years of update I have no hope that this will be any different.  Time to face reality, the board is junk.

Uup115, 2022-01-07:
> as of 1.4.5 rev2 is still a colossal failure imho. it drops and never starts again. I may try a mechanical timer to turn it off at night for an hour, who knows. there is a 1.4.7 but even with the [...]

1.4.8 still hangs.  I'm sweating bullets here.  I've spent nearly two weeks trying to circumvent this dreaded hang, with no success.  Please someone fix it.