arduino-libraries/WiFiNINA #50: uno wifi rev 2 firmware

icy711, 2019-03-08:
sorry if this is not the right place to post, but I updated the firmware and tried uploading the AP_SimpleWebServer example and it fails to create an access point so I uploaded the firmware check and it tells me to put on 1.2.0, so I went back to the firmwareupdater sketch and the firmware tool only shows 1.2.1
1.1.0
1.0.0

WiFiNINA firmware check.
Firmware version installed: 1.2.1
Latest firmware version available : 1.2.0
Check result: NOT PASSED
 - The firmware version on the module do not match the
   version required by the library, you may experience
   issues or failures.

if this is the wrong place to post let me know where is more appropriate that this team checks
Thanks
[image]

Rocketct, 2019-03-08:
@icy711 don't worry about the error, is the example that should be updated, the last version for the Uno wifi rev 2 is the 1.2.1!!

sandeepmistry, 2019-03-08:
@icy711 did you modify the password value in the secret tab to be 8 or more characters? What SSID value did you set?

icy711, 2019-03-10:
Thanks for the replies.  I did not change anything from the default example

[image]

I uninstalled and reinstalled the Arduino IDE along with the library and boards because it looks like the includes or mappings were maybe messed up from the output.  New to the Arduino world and this IDE is awkward to me so I'm not sure I got all the settings right or how to check the include paths.  When I hit upload I get this in the output

In file included from ..\Documents\Arduino\libraries\WiFiNINA\examples\AP_SimpleWebServer\AP_SimpleWebServer.ino:19:0:
..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\libraries\SPI\src/SPI.h:51:0: warning: "EXTERNAL_NUM_INTERRUPTS" redefined
 #define EXTERNAL_NUM_INTERRUPTS   NUM_TOTAL_PINS
 ^

In file included from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino/UART.h:28:0,
                 from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino/Arduino.h:139,
                 from sketch\AP_SimpleWebServer.ino.cpp:1:
..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\variants\uno2018/pins_arduino.h:40:0: note: this is the location of the previous definition
 #define EXTERNAL_NUM_INTERRUPTS     48
 ^

In file included from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\libraries\SPI\src\SPI.cpp:20:0:
..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\libraries\SPI\src\SPI.h:51:0: warning: "EXTERNAL_NUM_INTERRUPTS" redefined
 #define EXTERNAL_NUM_INTERRUPTS   NUM_TOTAL_PINS
 ^
In file included from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino/UART.h:28:0,
                 from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino/Arduino.h:139,
                 from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\libraries\SPI\src\SPI.h:23,
                 from ..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\libraries\SPI\src\SPI.cpp:20:
..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\variants\uno2018/pins_arduino.h:40:0: note: this is the location of the previous definition
 #define EXTERNAL_NUM_INTERRUPTS     48

cut some errors our for brevity.  I can add them back in if need be

.\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino\WInterrupts.c: In function 'detachInterrupt':
..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino\Arduino.h:127:44: warning: initialization discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define getPINnCTRLregister(port, bit_pos) ( ((port != NULL) && (bit_pos < NOT_A_PIN)) ? ((volatile uint8_t *)&(port->PIN0CTRL) + bit_pos) : NULL )
^

..\Arduino15\packages\arduino\hardware\megaavr\1.6.25\cores\arduino\WInterrupts.c:100:29: note: in expansion of macro 'getPINnCTRLregister'
     uint8_t* pin_ctrl_reg = getPINnCTRLregister(port, bit_pos);
                             ^
Sketch uses 12622 bytes (25%) of program storage space. Maximum is 49152 bytes.
Global variables use 374 bytes (6%) of dynamic memory, leaving 5770 bytes for local variables. Maximum is 6144 bytes.

Rocketct, 2019-03-11:
@icy711 if you don't set the SSID and the PASSWORD the example not works, to make works you have to set the SSID that the board show as name, and the password(should be at least 8 character long) to connect to the AP that you create on the board, for example testing the sketch by setting : 
```
#define SECRET_SSID "a"
#define SECRET_PASS "aaaaaaaa"
```
on my test the sketch works with the uno WiFi rev 2

icy711, 2019-03-12:
Thanks for the extra info!
Might be worth putting in the comments the requirements (Both ssid and pw can't be blank) and a default ssid and password like:
#define SECRET_SSID "WiFiNina"
#define SECRET_PASS "12345678"

That way the example works out of the box.
Thanks again, and keep up the good work!

closed 2019-03-12 by icy711