arduino-libraries/WiFiNINA #55: Lacking API documentation

VerkkuUkkeli, 2019-03-26:
The library is very frustrating to use due to its lacking documentation. The reference page http://www.arduino.cc/en/Reference/WiFiNINA fails to mention the existence of several functions. For example I needed to use the following functions, for which no documentation was provided or some function overloads were not mentioned:

* `WiFi.channel(wifiAccessPoint)`
* `WiFi.BSSID(wifiAccessPoint, bssid)`

aster94, 2019-03-31:
@facchinm is `WiFi.BSSID(wifiAccessPoint, bssid)` from the API or since it is not in the arduino reference it should change?

VerkkuUkkeli, 2019-03-31:
> @facchinm is `WiFi.BSSID(wifiAccessPoint, bssid)` from the API or since it is not in the arduino reference it should change?

As far as I know, it is the only way to show the BSSID of access points found through scanning without first connecting to them.

aster94, 2019-04-01:
Yep, i would like to understand if they didn t pubblished it because they think to change the API or it is just an inadvertence

facchinm, 2019-04-02:
@aster94 it's part of the official APIs (WiFi101 and original WiFi libraries had it too). We are working to add the pages in the reference soon :slightly_smiling_face:

SimonePDA, 2019-04-13:
Working on the update of the reference. Almost finished.

SimonePDA, 2019-04-15:
The APIs docs and the firmware pages have been properly updated.

aster94, 2019-04-20:
anyway this function is still not documented:
`WiFi.BSSID(wifiAccessPoint, bssid)`

also it would be a good idea to document that the `BSSID()` and `macAddress()` functions return an array that should be read from 5 to 0. This lack in the documentation brought to misunderstandings with the esp32 folks (which are not always happy to follow the API) and now in their core these function must be read from 0 to 5 ([source](https://github.com/espressif/arduino-esp32/blob/f8eebb5c3907928d100b3b9d7d0fb0256e1dffa7/libraries/WiFi/src/WiFiSTA.cpp#L546))

@SimonePDA and @facchinm woudl it be possible to update the reference with these information to prevent problem like this in the future?

closed 2019-04-23 by SimonePDA