arduino-libraries/WiFiNINA #57: adding getHostname

aster94, 2019-04-13:
It would be useful to have the possibility to get the hostname, rith now it is possible only to set it: https://www.arduino.cc/en/Reference/WiFiNINAsetHostname

other platform have already added it, like esp32:
https://github.com/espressif/arduino-esp32/blob/6dab3f6777b47e077fecd5b27f5f316664dd6096/libraries/WiFi/src/WiFiSTA.cpp#L570-L580

sandeepmistry, 2019-04-26:
Hi @aster94,

Is there a particular use case you had in mind for this? Do you want to query the hostname you set or get the default one?

aster94, 2019-04-27:
hi @sandeepmistry,

In my opinion it would be useful especially to get the default one. I don't know if the arduino boards comes out from the factory with all the same hostname but anyway at some point the user could change easily it, thanks to the `setHostname()` but later he may forgot it and he can't check easily the new name. The user have two options set a new one or use his mobile phone/computer to see it.

Also, if someone would like to be 100% sure that the set function worked a similar approach could be made:

```cpp
bool change_hostname (String name)
{
    WiFi.setHostname(name);
    if (name != WiFi.getHostname())
    {
        return false;
    }
    else
    {
        return true;
    }
}
```

at the end if it is not too much code/trouble i would suggest arduino staff to implement it. Since arduino boards rely on esp32 and they already implemented it it would be possible to take a look at what they did maybe

sandeepmistry, 2019-05-01:
The only concern with this is the default hostname depends on the firmware version loaded on the board.

We have two options:
1) add a new command to the firmware to retrieve the hostname, and only support this feature in the new firmware versions.

2) Make the get host name calculate the default version of the hostname based on the firmware version if one was not set by the user, otherwise store the previously set one as a variable.

I'm leaning towards options two, since it would be compatible with all versions of the firmware. Option 1 is more robust however.

sandeepmistry, 2019-05-01:
@facchinm any thoughts on this?

facchinm, 2019-05-02:
Option 2 looks fine for me, at least in the meantime. We can decide to switch to option 1 after some months / when the boards being produced will be flashed with the new FW

CyrilleHulin, 2021-04-23:
Hello, Some news about adding getHostname?
I will need it too. :p