arduino-libraries/WiFiNINA #58: Duplicate SSID not handled.

thorium-g, 2019-05-01:
I might be wrong. Just working a coupe of hours with WiFiNINA. 
This possible problem might be a severe security issue.
I tracked the issue back to wifi_drv.cpp. 

I noticed that if there are wifi networks with identical SSID available, for example "ABC123", the behavior of programs based on WifiNINA 1.40 might become unpredictable.

It looks like the developer of this lib thought that SSIDs are some sort of UUIDs.
But SSIDs are not unique. even on a single channel, e.g. 2.427GHz, multiple networks with the same SSID could be created.  

Long story short instead of begin SSID (+PWD) there should be at least an option to select the network, e.g. index, from the structure of detected/available networks during scan. 

Tonight I really have no time for layer 2 research, also the "mesh" technology of AVM is not open source and they might do some tricks with mac-address-tables (and the behavior of mesh nodes is pretty stupid...)

Here is a Log from scan networks
also see attached screenshoot.
--------------------------------------------------------------------
02:10:23.619 -> Name: 
02:10:23.619 -> ABC123
02:10:23.619 -> Signal: 
02:10:23.619 -> -58 dBm
02:10:23.619 -> ~~~~~~~~~~~~~Found a Home-SSID!
02:10:23.619 -> Name: 
02:10:23.619 -> ABC123
02:10:23.619 -> Signal: 
02:10:23.619 -> -73 dBm
02:10:23.619 -> ~~~~~~~~~~~~~Found a Home-SSID!
02:10:23.619 -> Name: 
02:10:23.619 -> ABC123
02:10:23.619 -> Signal: 
02:10:23.619 -> -75 dBm
02:10:23.619 -> ~~~~~~~~~~~~~Found a Home-SSID!
02:10:23.619 -> Name: 
02:10:23.619 -> ABC123
02:10:23.619 -> Signal: 
02:10:23.619 -> -88 dBm
02:10:23.619 -> ~~~~~~~~~~~~~Found a Home-SSID!"
-------------------------------------------------------------------------
[image]

sandeepmistry, 2019-05-01:
@thorium-g would you be able to share the sketch you are using?

thorium-g, 2019-05-01:
Sure. This sketch is based on "the nerd example" packed with wifinina 1.4.0. 

I further investigated and refreshed my knowledge about Layer 2 WiFi. LLC  LLC (Physical Addressing) (Hardware Address + BSSID) is typically used to uniquely identify wifi-networks. 

I´m a native perl not c coder and with the "missing" documentation in wifi_drv.cpp spi_drv.cpp I am not able to provide a fix by myself, sry.

[The_Nerd-lite.zip](https://github.com/arduino-libraries/WiFiNINA/files/3135633/The_Nerd-lite.zip)

thorium-g, 2019-05-01:
Sure. This sketch is based on "the nerd example" packed with wifinina 1.4.0.

I further investigated and refreshed my knowledge about Layer 1&2 WiFi. LLC (Physical Addressing) (Hardware Address + BSSID) is typically used to uniquely identify wifi-networks.
http://seat.massey.ac.nz/159334/lectures/week2_3_1s.pdf see page 26

I´m a native perl not c coder and with the "missing" documentation in wifi_drv.cpp spi_drv.cpp I am not able to provide a fix by myself, sry.

[The_Nerd-lite.zip](https://github.com/arduino-libraries/WiFiNINA/files/3135937/The_Nerd-lite.zip)

thorium-g, 2019-05-02:
@sandeepmistry If you are looking for a free and usable wifi-scanner I can recommend WifiInfoView by nirsoft.net. Other tools I used are inssider (needs registration now :/) and Netstumbler (supports only limited set of wifi chipsets).

Also I am looking for option to gather info and details from the network I am actually connected to. It looks like, from what I can tell from the sourcecode of wifinina, that some of the function are implemented in wifi_drv.cpp but not (yet) integrated into the wifinina interface.

sandeepmistry, 2019-05-13:
@facchinm any thoughts on this? Really I think we are stuck with what the esp-idf provides in terms of scanning ...

@thorium-g are the channels of the networks you are scanning the same? See the [ScanNetworksAdvanced.ino](https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/ScanNetworksAdvanced/ScanNetworksAdvanced.ino) example sketch, for example how to get the channel.

sandeepmistry, 2019-09-18:
Hi @thorium-g, are you still interested in this? Please see my comment above ...

mitteKhanh, 2021-01-22:
Hi! We are running into a similar problem. 
We DO have networks with similar SSID's and would like to be able to connect to a specific one (the one that has the best signal)
Being able to specify an ID in begin() rather than the SSID would be great, looking at the code I wouldn't know how to proceed. any help would be greatly appreciated thanks

iSpartaco, 2021-06-08:
Hello, I am also stucking into this problem.
Any way to choose which SSID to connect to in case of many network with the same SSID? As previously suggested, BSSID seems to be ok, otherwise SSID + channel.

Thank you

Modris-lat, 2022-03-24:
> Hi! We are running into a similar problem. We DO have networks with similar SSID's and would like to be able to connect to a specific one (the one that has the best signal) Being able to specify an [...]

Hello!
We have the same issue and we have purchased 10 devices already, but this issue is critical.
Is there any solution currently?

Help would be greatly appreciated thanks