arduino-libraries/WiFiNINA #59: Intermittent no socket available / status 255 unless I disable interrupts

DesktopMan, 2019-05-10:
I have a timer interrupt (Timer B0, 10 Hz) I use for some timekeeping and unless I disable interrupts before calling client functions I get intermittent (1% - 10% occurrence) no socket available / status 255. Example:

```
noInterrupts();

if (!client.connect(SERVER_ADDR, SERVER_PORT))
  return false;

client.println(request);
client.println(host);
client.println("Connection: close");
client.println();
client.stop();

interrupts();
```

Comment out the noInterrupts() call and the errors occur. Is this expected behavior? Is it a symptom of some other issue? I would except the library to disable interrupts internally when required like the standard Arduino library does.

sandeepmistry, 2019-05-13:
Hi @DesktopMan,

Do you have a full sketch to reproduce the issue? Also, what board are you using?

JakubValtar, 2019-05-31:
Hi @sandeepmistry,

I'm running into this issue in MKR WiFi 1010 when using I2S library, which needs to run its interrupt every `4096 us` to read the audio buffer (which can take `~100-150 us`). Here is the code which reproduces the issue using only `Adafruit_ZeroTimer`. I reliably get `WL_NO_MODULE` within 2 seconds. When this happens, `WiFiUDP` instances get broken and stop sending and receiving.

```C++
#include <WiFiNINA.h>
#include "Adafruit_ZeroTimer.h"

Adafruit_ZeroTimer zt3 = Adafruit_ZeroTimer(3);
void TC3_Handler() { Adafruit_ZeroTimer::timerHandler(3); }

void my_isr() { delayMicroseconds(150); }

void setup() {
  Serial.begin(9600);
  
  if (WiFi.status() == WL_NO_MODULE) { while (true); }
  if (WiFi.firmwareVersion() < WIFI_FIRMWARE_LATEST_VERSION) { while (true); }

  for (;;) {
    uint8_t status = WiFi.begin("ssid", "password");
    if (status == WL_CONNECTED) break;
    delay(3000);
  }

  // Run my_isr every 4096 us (~244 Hz)
  // ticks/s = 48,000,000 Hz / 1024 (prescaler)
  // ticks/us = ticks/s / 1,000,000
  // period = 4096 us * ticks/us = 192
  zt3.configure(TC_CLOCK_PRESCALER_DIV1024,   // prescaler
                TC_COUNTER_SIZE_16BIT,        // bit width of timer/counter
                TC_WAVE_GENERATION_MATCH_PWM  // frequency or PWM mode
                );
  zt3.setPeriodMatch(192, 0, 0);
  zt3.setCallback(true, TC_CALLBACK_CC_CHANNEL0, my_isr);
  zt3.enable(true);

  Serial.println("setup done");
}

void loop() {
  uint8_t status = WiFi.status();
  if (status != WL_CONNECTED) {
    Serial.print("status: "); Serial.println(status);
    delay(500);
  }
}

```

Rocketct, 2019-06-07:
hi @JakubValtar with your code seems that simply introducing a delay in the loop() like:
```c
void loop() {
  uint8_t status = WiFi.status();
  if (status != WL_CONNECTED) {
    Serial.print("status: "); Serial.println(status);
    delay(500);
  }
delay(1);
}
```
allow the code never get the status 255, i had test with a deay of 1000 ms, 100 ms and 1 ms and had and each time had work, could be a work around to solve the problem,  however if you share  the sketch with the I2S i could investigate deeply in the code in order to find a definitive solution and check if this solution work also in this case!!
@DesktopMan i need a sketch to understand how to reproduce the issue, could you share a simplified version of yours?

Rocketct, 2019-09-06:
closed due to lack feedback, please reopen the issue if you need @DesktopMan

closed 2019-09-06 by Rocketct