arduino-libraries/WiFiNINA #60: Initialization changes pinmode on GPIO 10 on Arduino UNO WiFi Rev2

DesktopMan, 2019-05-10:
Afaik none of the 0-13 GPIOs on the board are in use by the WiFiNINA module, however during initialization the pinmode of pin 10 is changed to input for some reason, and I don't know why.

facchinm, 2019-05-13:
Hi @DesktopMan ,
can you provide a minimal code that shows the problem, with the expected outcome? Thanks

DesktopMan, 2019-05-13:
Sure can.

Expected: No change in direction of GPIO 10 during WiFi.status() call
Observed: Direction of GPIO 10 is changed to input during WiFi.status() call

```
#include <Arduino.h>
#include <WiFiNINA.h>

void setup()
{
    // Serial
    Serial.begin(9600);
    Serial.println("Booting up...");

    // Configure GPIO 10 as output
    pinMode(10, OUTPUT);

    // Write direction value
    Serial.print("Direction of GPIO 10 before WiFi.status() is: ");
    Serial.println(PORTB.DIR & 2);

    if (WiFi.status() == WL_NO_MODULE)
    {
        Serial.println("Communication with WiFi module failed!");
        return;
    }

    // Write direction value again
    Serial.print("Direction of GPIO 10 after WiFi.status() is: ");
    Serial.println(PORTB.DIR & 2);
}

void loop()
{
    delay(1000);
}
```

facchinm, 2019-05-13:
Thank you for the perfect example!
The bug is related with old AVR compatibility (where SS pin needed to be driven high/low at boot to give the SPI peripheral a master/slave role).
This is not needed anymore, so the easiest solution is to remove this line https://github.com/arduino/ArduinoCore-megaavr/blob/2b039643ecd605c2905b80234ce67f520e6658f0/libraries/SPI/src/SPI.cpp#L48 .
The best fix would be getting rid of the SS pin altogether, since it's always software controlled.
I'll prepare a PR soon to address the issue entirely. Thank you very much!

DesktopMan, 2019-05-13:
Is this also the case for _uc_pinMosi and _uc_pinSCK, except they are set to output? Or do those refer to internal pins not part of the standard GPIOs so doesn't affect user code.

facchinm, 2019-05-14:
You are right, MOSI MISO and SCK are all internal (not exposed on the headers) pins so the other assignments are correct.

closed 2019-05-24 by facchinm