arduino-libraries/WiFiNINA #62: WiFi.setHostname not working with Arduino WiFi Rev 2

andrea-m1, 2019-05-28:
I tried to place it before and after the begin method but the setHostname method doesn't seem to have any effect when using Arduino WiFi Rev 2.
Can you please help?

Rocketct, 2019-06-05:
Hi @andrea-m1 which version of the firmware are you using?  could you share a sketch to reproduce the error?

andrea-m1, 2019-06-10:
The latest 1.2.1. 
`#include <WiFiNINA.h>
int status = 0;

void setup() {
  // put your setup code here, to run once:
  WiFi.setHostname("MyArduino");
}

void loop() {

  // attempt to connect to Wifi network:
  while (status != WL_CONNECTED) {
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin("SSID", "Password");

    // wait 10 seconds for connection:
    delay(10000);
  }

}`

Rocketct, 2019-06-10:
thanks @andrea-m1 i'm making some tests with the f/w debug enabled, seems that  the nina receive the command and call the API to set the host name, i'm following to investigate.
The last nina f/w version is 1.2.3 could you check with this one?

andrea-m1, 2019-06-10:
I cannot seem to find v1.2.3 in the Arduino ide. 
I am assuming I have to download and compile it?
Could you please share the step by step guide?

per1234, 2019-06-10:
@andrea-m1 the hourly build of the Arduino IDE has NINA firmwares up to 1.2.3. So you only need to download and use the hourly build of the Arduino IDE for the WiFi101/WiFiNINA Firmware Updater tool to have version 1.2.3 of the firmware. You can download the Arduino IDE hourly build here:
https://www.arduino.cc/en/Main/Software#hourly

Note that the hourly build is intended primarily for beta testing, so I'm not necessarily recommending that you switch to using it from your regular Arduino IDE version (though more beta testers are always welcome). You can just use the hourly build to install the firmware and then go back to using a release version of the IDE.

sandeepmistry, 2019-07-16:
This will be fixed in the next version of the NINA firmware, by the following change: https://github.com/arduino/nina-fw/pull/24

cc/ @Rocketct

Rocketct, 2019-09-06:
@closed via https://github.com/arduino/nina-fw/pull/24

closed 2019-09-06 by Rocketct