arduino-libraries/WiFiNINA #71: AccessPoint issues

albydnc, 2019-07-10:
Some issues with the wifi.beginAP() command:
### Default Espressif AP showing up
Using Wifi Firmware 1.1.0 and WiFi NINA library 1.4.0, the hotspot is created ignoring my config (ssid,pass), and being an open network with ssid like ESP-XXXXXX
### Update to firmware 1.2.1
When updated the wifi module to the last firmware available, the hotspot fails to initialize.

**NB** I'm using MKR WIFI 1010 & Arduino IDE 1.8.9, I've also tried with the Web Editor and the 1.8.8, no luck

albydnc, 2019-07-10:
Tried with a second MKR WiFi 1010, reporting same errors

facchinm, 2019-07-10:
Hi @albydnc ,
can you share the SSID and password you are trying to initialize? Please note that, since the AP connection is using WPA standard, the password must be at least 8 characters long or the initialization will fail.
Fw 1.1.0 AP configuration is known to fail; for the latest firmware you can check https://github.com/arduino/Arduino/pull/8976 (that also brings nice speed improvements)

sandeepmistry, 2019-09-18:
I'm closing this for now due to lack of feedback, please re-open with requested info if you are still interested.

closed 2019-09-18 by sandeepmistry