arduino-libraries/WiFiNINA #72: Expected latency for UDP?

kierenj, 2019-07-24:
Hi - I'm doing some tests using UdpClient, both multicast and on a regular port. I updated to the latest firmware yesterday.

There's some latency - looks like around 200ms as best I can tell, although it does vary a lot. I'm basically using (pseudocode-ish):

```c
WiFi.setHostname("blop1"); // this doesn't work, by the way - hostname is `arduino-1ce0`
WiFi.begin("my network", "pass..");
udp1 = WiFiUDP();
udp1.beginMulticast(IPAddress(226,1,1,1), 4096);
udp2 = WiFiUDP();
udp2.begin(4096);

void loop() {
  int d;
  d = udp1.parsePacket();
  if (d > 0) { /*  invert LED */ }
  d = udp2.parsePacket();
  if (d > 0) { /*  invert LED */ }
}
```

If I hit the send button on my laptop every second, it inverts the LED every second, after a delay.

If I spam the send button (4 per second say?), they seem to get batched or something.. it kind of tends to tend towards on or off, flickering to the other. If I slow down, it's even again.

Is that kind of latency expected? I'm hoping for <50ms, ideally..

kierenj, 2019-07-27:
In fact I get similar for ping:

`Minimum = 3ms, Maximum = 319ms, Average = 168ms`

Is this just inherent in the wifi chip? The arduino board (MKR WIFI 1010)?

kierenj, 2019-07-27:
Nope, I was just misled by the docs!

https://www.arduino.cc/en/Reference/WiFiNINANoLowPowerMode

...says that the default mode is "no low power mode".

It's wrong. If I run WiFi.noLowPowerMode(), I get 3-5ms latency.

kierenj, 2019-07-29:
(Also - the docs dont mention the multicast functions)

kierenj, 2019-07-29:
Ah I found another - if I listen for multicast UDP packets on a given port, I receive non-multicast packets sent to the unit's own IP address on the same port (not multicast packets and not sent to the multicast address).

Anyway - I'm getting there with my project now. Thanks for putting this all together.. some frustrations but I am grateful.

Rocketct, 2019-07-31:
hi @kierenj thank you for report, i'm investigate over all the reference that you have shared, i have just make a first run of tests, the results are the follow:

1.  can i ask to share also the tx side? in my test i have used two boards one that receive and one that transmit each 10 ms (the sketch are attached to this message) and using your configuration(both instantiated multicast and non-multicast) the results was:

-   setting the port 3000 for non-multicast and 4096 for multicast:
     `
  udp1.beginMulticast(IPAddress(226, 1, 1, 1), 3000);
  Serial.println("connected ted udp1 multicast");
  udp2.begin(4096);
`
i got two results: non-multicast: the receiver receive only in the object without the stranger behavior, the time are respected, without lag or delay. i have compared the time showed by my logger and a wireshark capture, and are both compliant;
multicast the time are shorter(i'll investigate deeply over this);
after a some iteration the Rx board  hang;

2. testing low power and noLowPower, the results(in order how i test it) was:
    -  without both the receiving time was 10 ms fpr each packet;
    -  calling noLowpower the time is respected;
    - calling lowpower the rx time increase, and the performance are reduced as expected
    - recalling noLowpower the normal behaviour is restored.
then seems the the default one is noLowPoer mode.

3. the `setHostname(...);`  will be fixed in 1.2.4 by https://github.com/arduino/nina-fw/pull/24

4. thank you about  about the library documentation report on multicast; 

5. i have also checked last case share(same port)  the moodule can use only one configuration at time then, it receive on multicast also the non packet on because you create the multicastobject as first, if you invert the two begin it receiveonly non multicast, i made the following tests:

- instance multicast first- read multicat after the non-multicast in loop- send to non multicast: received by multicast

- instance multicast first- read multicat after the non-multicast in loop - send to non multicast: received by multicast 

- instance non multicast first- read multicast after the non-multicast in loop - send to non multicast: received by non-multicast

- instance non multicast first- read multicast after the non-multicast in loop - send to non multicast: received by non-multicast

- instance non multicast first- read multicast after the non-multicast in loop - send to multicast: noit received

as how is implemetned the udp class you can made more instance but with different ports

sandeepmistry, 2019-09-18:
Hi @kierenj,

Do you have any feedback on @Rocketct's comment above?

In particular:
> can i ask to share also the tx side? in my test i have used two boards one that receive and one that transmit each 10 ms (the sketch are attached to this message) and using your configuration(both instantiated multicast and non-multicast) the results was: