arduino-libraries/WiFiNINA #81: Its compatible with Arduinos 33 ble ?

enrutador, 2019-09-01:
Great job! Is this library compatible with the new family of arduinos 33 Ble?

Thank you

per1234, 2019-09-01:
The WiFiNINA library is for the Arduino boards that have a u-blox NINA-W102 ESP32 WiFi module attached. The Nano 33 BLE boards don't have a NINA-W102 module so you can't use them with the WiFiNINA library. These boards do not have WiFi functionality, only BLE provided by the NINA-B306 module.

So the only way it could possibly make sense to even try using this library with a Nano 33 BLE would be if you had a separate NINA-W102 attached to your Nano 33 BLE.

If you want a Nano board with built-in WiFi, I recommend the Nano 33 IoT:
https://store.arduino.cc/nano-33-iot

enrutador, 2019-09-01:
Hi per1234 
A lot of thanks clarification ¡¡

closed 2019-09-01 by enrutador