arduino-libraries/WiFiNINA #85: WiFi.getTime() not in documentation

ocrdu, 2019-09-19:
Hi,

WiFi.getTime() is implemented in this library and works fine, but is not documented on https://www.arduino.cc/en/Reference/WiFiNINA.

thekunalsaini, 2020-02-28:
would you like to add documentation of WIFI.getTime()

rjs5327, 2020-04-17:
rtc.begin();

  unsigned long epoch;
  int numberOfTries = 0, maxTries = 6;
  do { epoch = WiFi.getTime();   numberOfTries++;  if(epoch == 0){delay(2000);} }
  while ((epoch == 0) && (numberOfTries < maxTries));

  if (numberOfTries >= maxTries) {  Serial.print("NTP unreachable!!");  }
  else {
    Serial.print("Epoch received: ");
    Serial.println(epoch);
    rtc.setEpoch(epoch+(GMT*3600));
    Serial.println();
  }

rjs5327, 2020-04-17:
This is tested and proven on MKR 1010

rjs5327, 2020-04-17:
per1234,

What is the bug?  I am running this code on a MKR1010 right now with no issues.  It just needs added to the documentation and i dont have access.

rjs5327, 2020-04-17:
[GetTime.zip](https://github.com/arduino-libraries/WiFiNINA/files/4494503/GetTime.zip)

per1234, 2020-04-17:
>What is the bug?

@rjs5327 the bug is that part of the public API of this library is undocumented. It's a defect in the documentation, thus the bug label.

>It just needs added to the documentation and i dont have access.

Yes, this library's documentation is not hosted in a public repository so only someone with edit access to Arduino's documentation is able to fix this. Since none of us are able to do anything about this issue, I suggest we refrain from any further discussion here, since it serves no constructive purpose. Commenting unnecessarily on issues spams the people watching the repository.

imranmoezkhan, 2020-12-21:
Apparently this is now documented:
https://www.arduino.cc/en/Reference/WiFiNINAGetTime
But isn't mentioned anywhere in wifi NINA's reference page:
https://www.arduino.cc/en/Reference/WiFiNINA

Is the reference page a hanging link?

karlsoderby, 2021-02-02:
This issue is now fixed. Thank you everyone for reporting it.

closed 2021-02-02 by per1234