arduino-libraries/WiFiNINA #86: Setting fixed IP address while keeping DHCP-supplied DNS, gateway, and subnetmask

ocrdu, 2019-09-19:
Hi,

Not an uncommon scenario (because it can be convenient): connect first with dhcp to get an IP adress, DNS, gateway, and subnetmask, and then change the IP address. I tried:

```
while (WiFi.status() != WL_CONNECTED) {
  WiFi.begin(ssid, pass);
  delay(5000);
}
 WiFi.config(IPAddress(192,168,1,99));
```

This does set the IP address, but removes the other adresses I got from the DHCP (both WiFi.gatewayIP() and WiFi.subnetMask() report 0.0.0.0 after calling WiFi.config() with just an IP address).

I tried to get around this by trying something like:

```
WiFi.config(IP, IPAddress(0,0,0,0), WiFi.gatewayIP(), WiFi.subnetMask());
```

That works fine, except for the DNS, of course; the library doesn't implement a WiFi.getDNS() or WiFi.dnsIP() or some such to get at the DHCP-assigned DNS address(es).

So, what I would like to see is a way to set a fixed IP address without discarding the other addresses received from DHCP, or, alternatively, a way to retrieve the DNS server addresses given out by the DHCP server.

If at all possible 8-).

Rocketct, 2020-02-27:
Hi @ocrdu, i take a look on this issue and i had try to make some tests, implementing my own version of the get DNS, the main problem is that when you use config, the dhcp is stoped from the nina firmware, than getdns could be implemented but useless when used with config, the best you can do is give a DNS server like the google's one 8.8.8.8  in order to resolve the name address

ocrdu, 2020-08-06:
What if ... Suppose you don't use config, so that DHCP keeps running on the NINA firmware. Is there a way of retrieving the IP address of the DNS server that was assigned by DHCP to the NINA?

As things are now, you can retrieve the localIP, the gatewayIP; and the subnetMask; would it be possible to implement a function WiFi.DNSIP() that retrieves the IP address of the DNS server that is being used by the NINA?

ocrdu, 2020-08-12:
I suppose the above would involve exposing esp_netif_get_dns_info() in the NINA-FW .

jandrassy, 2020-08-12:
use DHCP and reserve the IP address on the DHCP server. then the Arduino will have always the same IP address

ocrdu, 2020-08-12:
That requires access to the DHCP server, which I don't always have, and also does not give me a way to retrieve on the Arduino what DNS server is being used. By hand, I can do everything, but I prefer automation.

jandrassy, 2020-08-12:
> That requires access to the DHCP server, which I don't always have, and also does not give me a way to retrieve on the Arduino what DNS server is being used. By hand, I can do everything, but I prefer automation.

if the network is not in your control how do you know a not conflicting IP address?

ocrdu, 2020-08-12:
That requires access to the DHCP server, which I don't always (want to) have, it's a manual configuration thing (yuck 8-) and also does not give me a way to retrieve on the Arduino what DNS server is being used after a (DHCP or other) configuration.

I will close this (if I can) and make an enhancement request for adding a WiFi.getDNS() to the library.

closed 2020-08-12 by ocrdu