arduino-libraries/WiFiNINA #89: Problems with WiFiSSLClient

edwRoland, 2019-10-18:
Hi,

i have problems with connecting to a specific webside. 
Others work but the website "www.bayrol-poolaccess.de" doesn't.

I've updated the SSL root certificates from the said website and its root certificate provider: digicert.com

Is there any way how i can debug this? Error Messages or Debug Messages to help me what I'm doing wrong?

Thank you very much for your help and time!

Arduino 1.8.10
Arduino NANO 33 IoT
NINA firmware version 1.2.4 Arduino MKR WiFi 1010

sandeepmistry, 2019-11-14:
Hi @edwRoland,

I can reproduce this with my MKR WiFi 1010.

I've tried to connect with openssl, and it seems the server is sending a self signed certification:

```
$ openssl s_client -tls1_2 -connect  www.bayrol-poolaccess.de:443
CONNECTED(00000005)
depth=1 C = DE, ST = NA, L = H\C3\83\C2\BCttenberg-Weidenhausen, O = IEL Elektronik-Systeme GmbH, OU = NA, CN = luna287.startdedicated.de, emailAddress = mail@ielgmbh.de
verify error:num=19:self signed certificate in certificate chain
verify return:0
---
Certificate chain
 0 s:/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
   i:/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
 1 s:/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
   i:/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIEYjCCA0qgAwIBAgIDEAABMA0GCSqGSIb3DQEBBQUAMIG0MQswCQYDVQQGEwJE
RTELMAkGA1UECBMCTkExITAfBgNVBAcUGEjDvHR0ZW5iZXJnLVdlaWRlbmhhdXNl
bjEkMCIGA1UEChMbSUVMIEVsZWt0cm9uaWstU3lzdGVtZSBHbWJIMQswCQYDVQQL
EwJOQTEiMCAGA1UEAxMZbHVuYTI4Ny5zdGFydGRlZGljYXRlZC5kZTEeMBwGCSqG
SIb3DQEJARYPbWFpbEBpZWxnbWJoLmRlMB4XDTE1MDYyMjEwMTEyNFoXDTI1MDYx
OTEwMTEyNFowgbQxCzAJBgNVBAYTAkRFMQswCQYDVQQIEwJOQTEhMB8GA1UEBxQY
SMO8dHRlbmJlcmctV2VpZGVuaGF1c2VuMSQwIgYDVQQKExtJRUwgRWxla3Ryb25p
ay1TeXN0ZW1lIEdtYkgxCzAJBgNVBAsTAk5BMSIwIAYDVQQDExlsdW5hMjg3LnN0
YXJ0ZGVkaWNhdGVkLmRlMR4wHAYJKoZIhvcNAQkBFg9tYWlsQGllbGdtYmguZGUw
ggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDH5ckvKSgo4DOSfOOzAxiJ
uCIYo4IM67W3YvQI6sWC6pEXUr4ux5d4WP9YPMEEgIfdAoT6gi5uZu6aQJhB1gBN
5hJiNP5Q6/RFx8zx/2cfcAZsHI2DbN6gqd+hqmtMF5TFlx6Q8UsKFRzffzSSk5Po
iB83JxiVhna8OA3vTRmOVqNL/KMIXfFB9SvYPc29qaVbM5+9FZNHS0Gu/YDqBkTo
S1kD4oTy7QARbEzJsEDxwNh6oLt3NB9cC8wWSX7odGEnjUq95Gvdnw6WlRtI1OAM
iqqUR3miqIChxVZXd5u/GWX2ixDitTjdaMQIOsIGH56V6MLBD91ygTk9EcndjrYp
AgMBAAGjezB5MAkGA1UdEwQCMAAwLAYJYIZIAYb4QgENBB8WHU9wZW5TU0wgR2Vu
ZXJhdGVkIENlcnRpZmljYXRlMB0GA1UdDgQWBBQrKN/AtdwYFtNF48ifHFEP3ePE
xTAfBgNVHSMEGDAWgBQBzS+4rnW316RGMtdAdLXSPxzyzzANBgkqhkiG9w0BAQUF
AAOCAQEAAd5jveRX86lqfjQVuSApyS0GY/ggLeDMo7tV6ufwiAS9YzQlF3aqyVDf
XQgXD8oyCuZ5wGJp1Ej764ce5wU78VqCjytt7L6EeftWLwFowNz3VLC0RQgSRvlG
/DpAic3JeGAaXcvuUY/k1WoqNfpMy5jhRZ3jhoxFHjG6p51wCnFU9lXdakkE0X2D
O6ydwlem2443Yp0Argsxrh2mk2xM4xhOY9ri4NUYzKv8mg7wXd8f4IQR4e/jHUO/
GMtyWRljuYILczhZghjjmvOsh7V8qvTV1PgqFrdLnIcsAgiOtyuw2MTzzqoLKDIb
G423GE2N8ZGiKShuI78N6iNekzPsVw==
-----END CERTIFICATE-----
subject=/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
issuer=/C=DE/ST=NA/L=H\xC3\xBCttenberg-Weidenhausen/O=IEL Elektronik-Systeme GmbH/OU=NA/CN=luna287.startdedicated.de/emailAddress=mail@ielgmbh.de
---
No client certificate CA names sent
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 3111 bytes and written 326 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: 7BE8B9DA4901441498DECF55971347BB49D6CA1726F5C51D9BA31B13179B6963
    Session-ID-ctx: 
    Master-Key: 2715609C64B2CB6B7EF74A3356FA1588A042BB2306C841A28389332875207459012CD58B548325A9B1AC0BB86E34E67B
    TLS session ticket lifetime hint: 300 (seconds)
    TLS session ticket:
    0000 - 21 1a 8a 46 0d 07 98 96-1c 66 49 0e ae 6c ec 66   !..F.....fI..l.f
    0010 - 0c bd 64 e8 d1 3e 69 78-73 1e 29 d9 f8 89 99 60   ..d..>ixs.)....`
    0020 - 91 2f 30 f8 e3 47 6d 44-7e 5b a0 c0 16 19 09 5b   ./0..GmD~[.....[
    0030 - b7 35 47 cc ad e8 1b f8-34 b6 6d f3 4f 65 32 ff   .5G.....4.m.Oe2.
    0040 - 51 94 39 c8 17 1a 59 11-d2 87 b4 17 cc 37 79 9d   Q.9...Y......7y.
    0050 - 9e 7c d8 14 71 13 c4 6f-da 69 ed 59 a5 a7 9f a5   .|..q..o.i.Y....
    0060 - ce 82 a7 f6 6e 8a 36 df-bd ae 7b dc 15 9b 45 6a   ....n.6...{...Ej
    0070 - 6f 73 3b 2d 7d da 5f 98-be 1c b2 bf 4a ee 3a a8   os;-}._.....J.:.
    0080 - c6 2d 05 be cc 5c 65 5c-6a 91 9b 64 19 fd 4a 73   .-...\e\j..d..Js
    0090 - 04 4b 23 10 ea 00 c7 b5-47 5d bd 1f 9e 6c 81 95   .K#.....G]...l..
    00a0 - 10 63 df da a2 5a 7b 57-34 5c b7 32 11 db dd c2   .c...Z{W4\.2....
    00b0 - 46 d1 0d 2b c3 f2 22 df-36 29 83 61 be c6 9e 62   F..+..".6).a...b

    Start Time: 1573764251
    Timeout   : 7200 (sec)
    Verify return code: 19 (self signed certificate in certificate chain)
---
```

Would you be able to contact the servers administrator?

edwRoland, 2019-11-16:
Hi sandeepmistry,

thank you very much for having a look into this!

> Would you be able to contact the servers administrator?
Sadly not. Is there a way on allowing self signed certificates?

Regards,
edwRoland