arduino-libraries/WiFiNINA #92: Issues in WiFiNINA for MKR 1010 and Nano 33 IoT

per1234, 2019-11-15:
*Moved from https://github.com/arduino/Arduino/issues/9427 by @racpra16*

2 functions called setupNetwork(), you can change between AP & connecting to existing network

If the user connects to an existing network. It can reach the host using trace root and the hostname.

If a user creating an AP and connect to the create ap. It cannot reach, the Arduino using trace root and the hostname. But he can use an Arduino AP.

Parallelly, hostByName function works on the network but not on the AP. Also, the function setHostName does not seem to work on the Arduino nano IOT 33.

---
CC: @Russell108

Russell108, 2019-11-15:
ok I have 2 issues

1   hostByName function does not work on arduino created AP
2  setHostname does not seem to work on nano iot 33 or MKR 1010

Rocketct, 2019-11-15:
Could be related to the same issue but i need more details at this point.
i made this PR https://github.com/arduino/nina-fw/commit/f63b70aa3d4091003248f22bd1a176dfb9ce7e77 some time ago that fix the setHostName in our customer firmware for nina-W102, i can confirm that this not works when the board who should be assigned the host name is configured as access point (`WiFi.beginAP(ssid, pass);` to be clear), this patch is available only from 1.2.4 f/w version then please check too the f/w of the device, i'll check if works too on the iot 33.

please if i make confusion with the configuration, say to me.

About the access point issue would you exactly describe your experience and share a minimal sketch to reproduce it? please take a look also on the comment i made here if https://github.com/arduino/Arduino/issues/9427#issuecomment-554277812 to check my experience and eventually reply here your note and consideration

Russell108, 2019-11-15:
HI Riccardo

Thank you for your assistance.I apologise if I am frustrated and annoyed.
I have attached a zip file which explains my issues Please let me know if you require further details

For the time being I am giving my arduinos fixed ip addresses.
Another more serious issue is that the nano iot 33 disconnects from wifi every now & then 
Are you guys aware of this issue 

[from Russell.zip](https://github.com/arduino-libraries/WiFiNINA/files/3851026/from.Russell.zip)

Rocketct, 2019-11-15:
> Another more serious issue is that the nano iot 33 disconnects from wifi every now & then

if you observe that this happen under particular condition could be interesting investigate, would you like to open an issue with details, sketch, ap model we should split the issue in order to make atomic the fix.
we will take in to account also the other problems you rise in the other place @racpra16 will open soon, and i'll will assign to me!

thank you for fast reply!

Russell108, 2019-11-15:
Riccardo

I think it might be at my end I was powering from USB hub & I think when I put my computer to sleep the power outputs changed
I will close the tickets about power issues for the time

Thank you

Rocketct, 2019-11-18:
Hi @Russell108 little update, about:
>**Using Sketch setHostnameJoinNetwork**
>We now run the same tests but have added the function to set the host name
>At https://www.arduino.cc/en/Reference/WiFiNINAsetHostname
>
>This is what the arduino library example says
>WiF.setHostname(“MyArduino”); 
>
>I assume this is an error and should say
>WiFi.setHostname(“MyArduino”); 
>
>When I run this sketch, on the MKR 1010 using function
>1 setupJoinNetwork- connect to existing network
>the DHCP server still reports the host name as arduino-cc30
>
>I didn’t try it on the arduino iot 33
 
this could be a problem of firmware version, in the attached screenshots, i had adopt a man in the middle configuration were i set a raspberry as access point and captured the packets sent by both the boards (the iot dn the 1010 )with nina f/w 1.2.4.
As you can see the boards set the host name correctly, moreover the ping from the AP, how you can  see on the shell open in each picture, reach the end point.

[image]
[image]

The 1.2.4 will be available soon.

about this, instead:
[image]

i could say that: the code section in the nina f/w where the access point is  managed don't take in to account the set host name set by setHostName, than i have to find a way to test it and make it works if possible, i'm working on it and reply as soon as i get a results.

Russell108, 2019-11-18:
How do you get the 1.24 firmware
Onto the nano IOT 33? 
Using the currently available tools its not possible as I understand it

Rocketct, 2019-11-19:
yes actually the 1.2.4 it's not available on the IDE's tool, and also not support the IoT33 but you should just have uploaded on the IoT33, should be the factory setting of the iot you could check the firmware version using the example:  `Example->WiFiNINA->tools->CheckFirmwareVersion`

Russell108, 2019-11-19:
In relation to

 “this could be a problem of firmware version, in the attached screenshots, i had adopt a man in the middle configuration were i set a raspberry as access point and captured the packets sent by both the boards (the iot dn the 1010 )with nina f/w 1.2.4.
As you can see the boards set the host name correctly, moreover the ping from the AP, how you can see on the shell open in each picture, reach the end point.”

So if this is the case I don’t understand how you can run 1.2.4 on the Nano IOT 33 as I understood you stated in your previous communication?
If I run firmware checker on my iot Nano 33 says 1.2.3
Thanks

Rocketct, 2019-11-19:
>If I run firmware checker on my iot Nano 33 says 1.2.3

Ok this confirm that you are not able to set the host name because this feature on that version is not supported.

> should be the factory setting

sorry rectify the factory is the 1.2.3, than i'll ask when the new version will be available.

>So if this is the case I don’t understand how you can run 1.2.4 on the Nano IOT 33 as I understood you stated in your previous communication?

I have updated the firmware with another procedure, there is a way to upload the firmware also from a shell tools and not only the IDE, but you have to use esptools (its a binary that you can download and run from the shell) and the binary of the firmware:

[NINA_W102.bin.zip](https://github.com/arduino-libraries/WiFiNINA/files/3862781/NINA_W102.bin.zip)

you can also compile the f/w through espressif's IDF (v. 3.3), you'll find  the install procedure  on the second link, if you want try you can, but please read te note in caps at the end of this comment, following the links you need:

- Pass through required to send the f/w on too the MCU( the module not communicate directly with the USB) : https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino;
- NINA f/w (i have just shared, this is the link to the source code that should be compiled, the instruction to install the tool are on the read me): https://github.com/arduino/nina-fw

**ALL THESE FEATURE ARE NOT SUPPORTED BY US, SO WE CANNOT SUPPORT YOU SORRY, MY SUGGESTION IS TO WAIT THE RELEASE OF SUPPORT FOR THE FIRMWARE VERSION AND THE BOARDS YOU NEED**

Russell108, 2019-11-29:
Hi Ricardo

Regarding the update to 1.2.4

I am finding the Nano IOT 33 is disconnecting from wifi after a certain period 1-6 hours

I think this used too happed with the MKR1010 and was resolved in 1.24 
Please can you tell me

Are you aware off this issue
Is this issue in the nano probably the same issue as  in the MKR
Any idea when the official release of the updater for the nano IOT 33 will be available

I need to use the boards in a live situation and this is causing a problem for me

Thanks Russell

Russell108, 2019-11-29:
Hi Riccardo

Regarding the pass through sketch

It appears it is written only for the MKR boards

  SerialNINAPassthrough - Use esptool to flash the u-blox NINA (ESP32) module
  Arduino MKR WiFi 1010, Arduino MKR Vidor 4000, and Arduino UNO WiFi Rev.2.

I don’r see an option  for the nano IOT 33 when reading the sketch

Can I still use this sketch for pass through

Rocketct, 2020-01-22:
hi @Russell108 you can try with the new firmware is available in the HOURLY BUILDS build of the IDE to the following link: https://www.arduino.cc/en/main/software

to check your connectivity issue as usual i need a sketch to reproduce it.

Russell108, 2020-01-23:
The hourly builds don’t work on Catalina

Russell108, 2020-01-23:
Re sketch I cannot open the arduino ice it says there is a permissions error to a certain folder

Russell108, 2020-01-23:
Ok

I just downloaded the latest update 23 18 : 24:38 Got

This does open on Catalina I will see if the wifinina updater works for me

Thank you

> hi @Russell108 <https://github.com/Russell108> you can try with the new firmware is available in the HOURLY BUILDS build of the IDE to the following link: https://www.arduino.cc/en/main/software [...]
Jan

Rocketct, 2020-01-24:
welcome @Russell108  let me know if the issue is still there, and we can follow on try to resolve it

Russell108, 2020-01-24:
Yes 

Now the builds are compatible with Catalina all is good thanks

Rocketct, 2020-02-26:
welcome!! i'll close the issue!!

closed 2020-02-26 by Rocketct

Russell108, 2020-02-26:
Thank you for fixing this
I have not yet implemented in my project as I did a work around but I am looking forward to doing so

Sorry I didn’t;’t acknowledge before

Thank you
Please have a great day