arduino-libraries/WiFiNINA #96: Up date firmware on Nano IOT 33

Russell108, 2019-12-16:
Unable to update firmware on Nano IOT 33, which will fix some bugs I have discussed elsewhere with the arduino team.

They suggested I could use third party tools to update but recommended I wait.
I have also been told the updater has been built to enable this but has yet to be posted in the main download of arduino so when can we expect this to appear.

Sorry if this is in the wrong place Its a wifi type issue so I posted here rather than the main arduino. Thanks

Russell108, 2019-12-17:
anyone?

sandeepmistry, 2019-12-23:
The following pull request in the IDE will provide NINA firmware v1.3.0: https://github.com/arduino/Arduino/pull/9552

Changelog for firmware changes can be found here: https://github.com/arduino/nina-fw/blob/master/CHANGELOG

closed 2019-12-23 by sandeepmistry

johnpneumann, 2020-01-07:
1.3.0 does not appear to be available for my Nano 33. The latest I see is 1.2.4 and 19.6.1.

Russell108, 2020-01-07:
Ok I updated the library to 1.5 
I open WIFININA firmware updater I see the nano IOT 33 board 
I RUN THE firmware check
WiFiNINA firmware check.
22:33:49.910 -> 
22:33:50.691 -> Firmware version installed: 1.2.3
22:33:50.691 -> Latest firmware version available : 1.2.1
22:33:50.691 -> 
22:33:50.691 -> Check result: PASSED
how to update the firmware

sandeepmistry, 2020-01-07:
You'll have to download the hourly Arduino build: https://www.arduino.cc/en/Main/Software#hourly

Russell108, 2020-01-07:
I get error message 

If I click ok or don’t allow the message just re appears

Russell108, 2020-01-07:
I don’t know if this is an issue

Russell108, 2020-01-08:
I cannot open the hourly build referenced or the nightly build on OS X Catalina
the operating system  does not allow access to folders.
seems like the image  attached to the email didn't arrive so here they both are
[image]
[image]