arduino-libraries/WiFiNINA #98: Add WiFiNINA WIFI.status() return values to arduino docs

rickstaa, 2019-12-18:
Maybe it is an idea to add the return values corresponding to the different WIFI.status() statuses to the [arduino documentation](https://www.arduino.cc/en/Reference/WiFiNINAStatus)? Otherwise, I have added them to this issue as a reference.

 **WiFiNiNa return values:**

-   0: WL_IDLE_STATUS
-   1: WL_NO_SSID_AVAIL
-   2: WL_SCAN_COMPLETED
-   3: WL_CONNECTED
-   4: WL_CONNECT_FAILED
-   5: WL_CONNECTION_LOST
-   6: WL_DISCONNECFTED
-   7: WL_AP_LISTENING
-   8: WL_AP_CONNECTED

SimonePDA, 2019-12-19:
Hi, I checked the WIFI.status() page of the documentation and it already reports all the returned statuses:

WL_CONNECTED: assigned when connected to a WiFi network;
WL_AP_CONNECTED : assigned when a device is connected in Access Point mode;
WL_AP_LISTENING : assigned when the listening for connections in Access Point mode;
WL_NO_SHIELD: assigned when no WiFi shield is present;
WL_NO_MODULE: assigned when the communication with an integrated WiFi module fails;
WL_IDLE_STATUS: it is a temporary status assigned when WiFi.begin() is called and remains active until the number of attempts expires (resulting in WL_CONNECT_FAILED) or a connection is established (resulting in WL_CONNECTED);
WL_NO_SSID_AVAIL: assigned when no SSID are available;
WL_SCAN_COMPLETED: assigned when the scan networks is completed;
WL_CONNECT_FAILED: assigned when the connection fails for all the attempts;
WL_CONNECTION_LOST: assigned when the connection is lost;
WL_DISCONNECTED: assigned when disconnected from a network;

https://www.arduino.cc/en/Reference/WiFiNINAStatus

May I ask where issue was found?

thanks

simone

SimonePDA, 2019-12-19:
@sandeepmistry might help in this giving me the info to add to the documentation..

akash73, 2019-12-19:
@rickstaa thanks for suggesting this but  the result is already a  constant and adding a number  does not add relevant info.
I'm closing this, thanks!

closed 2019-12-19 by akash73

rickstaa, 2019-12-20:
@akash73 Thanks for your response. I understand I will keep this issue as a reference then.

j2kun, 2020-12-01:
The number is helpful because if you are running Wifi.begin() and it's not connecting properly, you will Serial.print the status and then have to look it up from the number code alone.

mpierucci, 2021-05-08:
> The number is helpful because if you are running Wifi.begin() and it's not connecting properly, you will Serial.print the status and then have to look it up from the number code alone.

Exactly! I had to go through the code to understand the status I was getting. BTW I'm getting 9 ( when trying the `AP_SimpleWebServer`) couldn't find 9 in the source code, what does it means

Secoashimself, 2022-01-14:
> 
> Exactly! I had to go through the code to understand the status I was getting. BTW I'm getting 9 ( when trying the `AP_SimpleWebServer`) couldn't find 9 in the source code, what does it means

I had the same problem. I finally solved it with a password which was longer than 8 characters.  [(see documentation of beginAP())](https://www.arduino.cc/en/Reference/WiFiNINABeginAP)

rickstaa, 2022-05-03:
@jordelsonjunior Sorry, I am not sure if I understand your question. I do not know what the numerical value of `WL_NO_MODULE` is. That option was not there when I used the library. If you want to know, I think it is best to inspect [the code](https://github.com/arduino-libraries/WiFiNINA) found in the `master` branch.