arduino-libraries/WiFiNINA #103: WiFi Connection not Stable on NANO 33

Ravenbs, 2020-02-26:
Hello,

See also following Thread in Arduino Nano 33 Forum.
The WiFi connection is not stable and there is a reconnect neccessarry all few hours.

See Details here:
https://forum.arduino.cc/index.php?topic=664834.0

mtz8302, 2020-03-03:
I have the same issue. Any solutions?

Ravenbs, 2020-03-03:
I get no reactions from a developer or from Arduino on this, also not in the Arduino forum. Only chance is to reach out to amazon and send the nano 33 back. Think they just dont work...

Rocketct, 2020-03-06:
@Ravenbs i'm working on this i let run a nano 33 iot without dscisconnection from the AP for 2 days, now i'm planning to make more complex test

Ravenbs, 2020-03-06:
Thank you for this Feedback and your time!

Rocketct, 2020-03-06:
welcome i'll replay as soon as il'll get other results

mtz8302, 2020-03-06:
Thanks for your work. The code I'm running disconnects sometimes,  I don't know under which conditions, sometimes it works for a day, sometimes for  2h. On esp32 no probs 

https://github.com/mtz8302/AOG_Section_control_ESP32_Nano33iot

roberthartung, 2020-03-13:
I am trying to send/receive UDP Multicast packets. I see different problems after a few minutes/hours even:

- Random disconnects (rarely)
- `udp.beginPacket()` returns 65533 bytes (no idea why!)
- udp.read() does return the full packet most of the time, but sometimes it doesn't. Afterwards all packets are corrupted.

roberthartung, 2020-03-13:
And another error just happend. Node:

```
07:37:50.743 -> new packet: 19
07:37:50.743 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xE0 0x13 0x2 0xFA 0x3 0x3 0x0 0x80 0xE 0x8E
07:37:50.743 -> new packet: 17
07:37:50.743 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0xA 0x1 0x0 0x40
07:37:50.848 -> new packet: 17
07:37:50.848 -> Skipped byte: 2
07:37:50.848 -> Skipped byte: 8C
07:37:50.882 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0xB 0x1 0x0 0x40
```

tcpdump:

```
07:37:50.787915 IP (tos 0x0, ttl 255, id 988, offset 0, flags [none], proto UDP (17), length 47)
    10.1.1.140.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 19
        0x0000:  4500 002f 03dc 0000 ff11 b548 0a01 018c  E../.......H....
        0x0010:  e000 170c 0e57 0e57 001b d127 0610 0530  .....W.W...'...0
        0x0020:  0013 2900 bce0 1302 fa03 0300 800e 8e    ..)............
07:37:50.790109 IP (tos 0x0, ttl 255, id 989, offset 0, flags [none], proto UDP (17), length 47)
    10.1.1.140.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 19
        0x0000:  4500 002f 03dd 0000 ff11 b547 0a01 018c  E../.......G....
        0x0010:  e000 170c 0e57 0e57 001b d332 0610 0530  .....W.W...2...0
        0x0020:  0013 2900 bce0 1302 fa04 0300 8002 8c    ..)............
07:37:50.801504 IP (tos 0x0, ttl 255, id 1676, offset 0, flags [none], proto UDP (17), length 45)
    10.1.1.40.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 17
        0x0000:  4500 002d 068c 0000 ff11 b2fe 0a01 0128  E..-...........(
        0x0010:  e000 170c 0e57 0e57 0019 f157 0610 0530  .....W.W...W...0
        0x0020:  0011 2900 bcc0 1264 ab0a 0100 4000       ..)....d....@.
```

The first packet (19 byte) is received correctly.
The second packet should be 19 bytes as indicated by tcpdump, but is received as 17 bytes.

I wonder if the UDP Checksum of the original 19 byte packet was wrong, but the data was not discarded? Afterwards the next packet with 17 bytes is received, and then the old data is reported with the new length?!

mtz8302, 2020-03-22:
used the nano33iot with firmware 1.3 and changed code, but didn't help, as the reconnection takes too long. So I need a stable connection.
Did you find anything?

Rocketct, 2020-03-23:
No, in the counter check with other boards seems all fine, have you the possibility to use another AP? only to be sure that is not dependent from Fritz Box 7490.

Ravenbs, 2020-03-23:
From me also tested with Fritz Repeater 1200 without a difference. But its the same manufacturer have nothing else on hands.

luigigubello, 2020-03-23:
Hi @mtz8302, what is the router's model/vendor to which your Arduino Nano 33 IoT connects?

luigigubello, 2020-03-24:
Hi @Ravenbs, is your AP a FritzBox? If not, what is the model/vendor of your access point? Can you try to create another AP and connect your Arduino to this one? (e.g. you can use your phone) Thank you!

Ravenbs, 2020-03-24:
Yes I tested with the named Fritz Box. 
I not longer have a Nano 33 to test, sorry.

mtz8302, 2020-03-25:
> Hi @mtz8302, what is the router's model/vendor to which your Arduino Nano 33 IoT connects?

Hi I used it last week to cintrol my sprayer. It was loggt to Accespoit created by ESP32, that tranfsers GPS NMEA data to WIN10 tablet.
On my other tractor I have a teltonica RUT955. By the next wekk I will try this.

Testing it in the barn everything works. In the field there might be different conditions, so the UDP packets are not that regual in time and this may cause trouble.
Additionally I changed the code to keep it running more fluently.

I'll report if I've tested it in working conditions.

luigigubello, 2020-03-27:
Thank you! I have tried to reproduce the bug, but I cannot. About Fritz Box router my theory is a deauthentication issue, but I haven't a Fritz Box to test it.

roberthartung, 2020-03-27:
I had the same issues with a Unifi nano HD AP.

Ravenbs, 2020-03-27:
@roberthartung 
Can you give more detail about your Access point? Model number?

roberthartung, 2020-03-27:
@Ravenbs I already told you that. It's a [Unifi nano HD](https://unifi-nanohd.ui.com/).

R0BER, 2020-05-25:
I can confirm this error with mkr1010 ...
The same code with mkr1000 works perfectly for years.

When the problem appears pin 5 behaves strange, setting its level high.

router model RTF8115VW

vindea, 2020-08-31:
Hello,
I have the same problem with a nano 33 IOT and a Netgear Orbi wifi router.
The Nano is having difficulty connecting to wifi. And when it is connected, after about ten hours, the connection deteriorates. If I ping the Nano, 80% of the packets are lost. I am therefore obliged to reset the card.
Arduino Nano 33 IOT with Wifinina, firmware 1.3.0.

IgorSchelle, 2020-12-30:
I have the same problem with Arduino wifi MKR  1010.  I use Arduino online scatch editor with a pre-generated connection code.

LukeScrewdriver, 2021-04-27:
Hello,
I have the same problem too with arduino nano 33 ioT.

Trier71, 2022-04-10:
Adding myself here, with the same problems. Anybody out there who thought about modifying a libary to influence the re-connect behavior of the Nano33IoT?