arduino-libraries/WiFiNINA #105: Unable to connect to WPA2 Enterprise network.

ktbaker, 2020-03-02:
Hi there,

My university runs a WPA2 Enterprise network and I've been trying to connect my Arduino Uno Wifi Rev2 to it using the beginEnterprise() function. The authentication of the network is PEAP/MSCHAPV2. I've updated my firmware, tried powering the Arduino with an external power source, and tried many different credentials. My board connects to regular networks using the function begin() so it's also not a hardware issue. I've been using the example code sketches provided by the library to avoid errors in my code. When I run the ScanNetwork sketch the Arduino sees the network I'm trying to connect to and it lists it's Encryption as Unknown.

Please let me know if there is something else I should try.

Thanks,
Katie

wiegerthefarmer, 2020-03-03:
@sandeepmistry Any thoughts? I see you created the WPA2 Enterprise support.

facchinm, 2020-03-10:
Hi Katie,
I'm a bit bothered by the `Unknown` encryption on scan. How does the network show up on a laptop / smartphone? We tested extensively on [eduroam](https://www.eduroam.org) network but in that case the encryption field is always `Open`

ktbaker, 2020-03-12:
Hi, I'm a little confused by your question. On my laptop and smartphone the network shows up just like how any other network shows up? In testing, I was trying to connect to eduroam network as well and its' encryption was also unknown.

facchinm, 2020-03-13:
Sorry for the not so clear question, I was referring to something like this.  
[image]
Is the board working in the `eduroam` network you tested?

ktbaker, 2020-03-18:
I am on Apple devices so this information is not available but the security is 802.1x PEAP as per the university's website. The board does not connect to eduroam network.

invalidflaw, 2020-08-06:
Used my smartphone on my universities eduroam, it returns WPA2/WPA3-Enterprise. my nano-33 IOT cannot connect. Scanninng wifi returns unknown encryption.

khoih-prog, 2020-08-06:
So, you have to solve the problem, then connect manually and successfully before moving forward.

lennartfox, 2021-09-27:
Hi, I have a similar issue. The Arduino Nano 33 IoT wont connect to eduroam (WPA2 Enterprise) und is stuck in attempting to connect. When I run the ScanNetwork sketch the Arduino sees eduroam and it lists it's encryption as unknown. I have tried the credentials on a different device and the Arduino on a different WiFi. So it must be an error in the code. Could somebody clarify the defference between identity and user?

Furthermore, I have included the Certificate our webmaster gave me like so: 
`const static char caCert[] PROGMEM = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDojCCAoqgAwIBAgIQT8JKneArxqhO9YcYVsfWojANBgkqhkiG9w0BAQsFADBQ\n" \
//shortened for forum
"PlEXf5SN16SYY/Nb+WMKfRb/14YZHrfcsv6xbpQT7OqrvE+vdJA+GwtvGzmXHC40\n" \
"DAbgqAjSbfmy8s+DRL9ARmraStBA4g==\n" \
"-----END CERTIFICATE-----\n";`

Other than that, I use the example sketch. Does somebody now Why it is not working? I would appreciate your help.

Martino mentioned above, that you _tested extensively on [eduroam](https://www.eduroam.org) network_. Do you have any example sketch for that?

forcecodema, 2022-01-24:
Hi, I also have the problem with Arduino MKR WiFi 1010 (NINA Firmware 1.4.8) and connecting to WPA2 Enterprise network.
The following example code just runs forever. I can successfully gain access to the network using Apple and Android phone or Windows 10 PC. I am happy to provide some more info If that would help...

```
while (status != WL_CONNECTED) {
    Serial.print("Connecting to network: ");
    Serial.println(ssid);
    status = WiFi.beginEnterprise(ssid, user, pass);
    delay(10000);
}
```

CJacob314, 2022-10-27:
> Hi Katie, I'm a bit bothered by the `Unknown` encryption on scan. How does the network show up on a laptop / smartphone? We tested extensively on [eduroam](https://www.eduroam.org) network but in that case the encryption field is always `Open`

I am having the exact same issue, and both the `ScanNetworks.ino` and `ScanNetworksAdvanced.ino` code also give encryption types of "Unknown" for eduroam, and "None" for all non-enterprise networks. I wanted to add, however, that on both the ScanNetwork example code files, which can be seen [here](https://github.com/arduino-libraries/WiFiNINA/blob/master/examples/ScanNetworks/ScanNetworks.ino) for example, there is a comment which says:

```c++
/*
 This example prints the board's MAC address, and
 scans for available WiFi networks using the NINA module.
 Every ten seconds, it scans again. It doesn't actually
 connect to any network, so no encryption scheme is specified.
...
*/
```

This seems to imply that no encryption will be given for any networks as the chip does not actually attempt to connect to any of them. 

Has anyone found any solutions to this problem? My Arduino Uno Wifi Rev2 connects just fine on my home WiFi, but I would like to have it connect to my University WiFi as well.

ChathuZ, 2023-01-12:
> I am having the exact same issue, and both the `ScanNetworks.ino` and `ScanNetworksAdvanced.ino` code also give encryption types of "Unknown" for eduroam, and "None" for all non-enterprise networks. [...]

Did you figure it out? My university told me that "Connecting requires the device to support WPA2-Enterprise encryption for security reasons." I'm using a nano Iot and the network scan shows my WifI but the encryption as unknown.

CJacob314, 2023-01-12:
I was not able to figure out how to get my Uno WiFi Rev2 to connect to any WPA2-Enterprise networks.

I can potentially offer a small bit of advice, though. I thought I couldn't connect to my iPhone's hotspot because it failed when I typed out the SSID by hand. 

The issue, for me, was that my iPhone was using ASCII to represent all but one character, the apostrophe, which was being represented by three bytes in who knows what kind of Unicode. 

To solve this, I ran a modified example scan sketch, which would check if a network had a substring of my name, then wrote that string to my device's EEPROM with EEPROM.h. Then I could connect anytime by reading that string from the on-board ROM.

I used [these](https://roboticsbackend.com/arduino-write-string-in-eeprom/) functions.

ChathuZ, 2023-01-13:
Thank you CJacob314,
I spent couple of hours trying to figure out and this was very helpful. Tried what you said but I still couldnt figure it out because I'm a newbie to this. I changed my phones name without any special characters and Whola! it connected to the hot spot :-)

I still need to figure out a way to connect to my university Wifi because this is being built for an automatic data logger from a couple of sensors and I'm trying to water my plants automatically. I'm not sure if nano Iot supports the encryption level that my university's wifi requires. Maybe I will need to find a better board that supports it if this doesn't work. I can't afford a wireless modem with crazy data monthly premiums. 

Anyways, thank you for the help, really appreciate it.

Brexei, 2025-03-21:
I have the same problem and have not found a solution yet.
Our system administrator looked into the problem and concluded that WiFi.beginEnterprise routines are incomplete and they are not sufficient to connect to a pwa2 enterprise wifi network. The routines are just a few lines of code, but wpa2 enterprise connection requires much more and has two levels of validation.
As i am not a programmer, i cannot verify this. But please take a look and see if the code in the library is the proper one for connecting to such networks. Thank you