arduino-libraries/WiFiNINA #108: WifiUdp returns corrupt data / wrong packet length

roberthartung, 2020-03-13:
I am trying to send/receive UDP Multicast packets. I see different problems after a few minutes or hours:

- `udp.beginPacket()` returns 65533 bytes (no idea why!)
- udp.read() does return the full packet most of the time, but sometimes it doesn't. Afterwards all packets are corrupted.

```
06:52:08.852 -> new packet: 19
06:52:08.852 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0x14 0x3 0x0 0x80 0x75 0xB9
06:52:08.852 -> 1
06:52:09.235 -> BMESensor
06:52:09.270 -> new packet: 19
06:52:09.270 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xE0 0x13 0x2 0xFA 0x1 0x3 0x0 0x80 0xC 0x42
06:52:09.305 -> new packet: 19
06:52:09.305 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xE0 0x13 0x2 0xFA 0x2 0x3 0x0 0x80 0x6C 0xC4
06:52:09.305 -> new packet: 19
06:52:09.305 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xE0 0x13 0x2 0xFA 0x3 0x3 0x0 0x80 0xE 0x70
06:52:09.305 -> new packet: 18
06:52:09.305 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x12 0x29 0x0 0xBC 0xC0 0x12 0x2 0x72 0xD 0x2 0x0 0x80 0x38
06:52:09.305 -> 4
06:52:09.479 -> *new packet: 18*
06:52:09.479 ->   remaining: 17
06:52:09.479 -> Packet: 0x78 0x6 0x10 0x5 0x30 0x0 0x12 0x29 0x0 0xBC 0xC0 0x12 0x2 0x8A 0xD 0x2 0x0 0x80
06:52:09.479 -> WiFiKNX::parse - Error Header Length
06:52:09.479 -> 1
06:52:09.898 -> new packet: 19
06:52:09.898 ->   remaining: 18
06:52:09.898 -> Packet: 0x14 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0x14 0x3 0x0 0x80 0x75
06:52:09.898 -> WiFiKNX::parse - Error Header Length
06:52:09.898 -> 1
06:52:10.491 -> new packet: 17
06:52:10.491 ->   remaining: 16
```

Debugging it leads me to this:

```
06:52:09.305 -> new packet: 18
06:52:09.305 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x12 0x29 0x0 0xBC 0xC0 0x12 0x2 0x72 0xD 0x2 0x0 0x80 0x38
06:52:09.479 -> new packet: 18
06:52:09.479 ->   remaining: 17
06:52:09.479 -> Packet: *0x78* 0x6 0x10 0x5 0x30 0x0 0x12 0x29 0x0 0xBC 0xC0 0x12 0x2 0x8A 0xD 0x2 0x0 0x80
06:52:09.479 -> WiFiKNX::parse - Error Header Length
```

tcpdump on the network:

```
06:52:09.306125 IP (tos 0x0, ttl 255, id 61507, offset 0, flags [none], proto UDP (17), length 46)
    10.1.1.40.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 18
        0x0000:  4500 002e f043 0000 ff11 c945 0a01 0128  E....C.....E...(
        0x0010:  e000 170c 0e57 0e57 001a e97b 0610 0530  .....W.W...{...0
        0x0020:  0012 2900 bcc0 1202 720d 0200 8038       ..).....r....8
06:52:09.373948 IP (tos 0x0, ttl 255, id 61508, offset 0, flags [none], proto UDP (17), length 46)
    10.1.1.40.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 18
        0x0000:  4500 002e f044 0000 ff11 c944 0a01 0128  E....D.....D...(
        0x0010:  e000 170c 0e57 0e57 001a d19f 0610 0530  .....W.W.......0
        0x0020:  0012 2900 bcc0 1202 8a0d 0200 8014       ..)...........
```

For some reason there is an additional 0x78 in the packet on the Nano 33.

Afterwards I tried to look at the first byte, as it should be `0x06` all the time. Therefore I now skip bytes at the beginning.

And another error just happend. Node:

```
07:37:50.743 -> new packet: 19
07:37:50.743 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xE0 0x13 0x2 0xFA 0x3 0x3 0x0 0x80 0xE 0x8E
07:37:50.743 -> new packet: 17
07:37:50.743 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0xA 0x1 0x0 0x40
07:37:50.848 -> new packet: 17
07:37:50.848 -> Skipped byte: 2
07:37:50.848 -> Skipped byte: 8C
07:37:50.882 -> Packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xC0 0x12 0x64 0xAB 0xB 0x1 0x0 0x40
```

tcpdump:

```
07:37:50.787915 IP (tos 0x0, ttl 255, id 988, offset 0, flags [none], proto UDP (17), length 47)
    10.1.1.140.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 19
        0x0000:  4500 002f 03dc 0000 ff11 b548 0a01 018c  E../.......H....
        0x0010:  e000 170c 0e57 0e57 001b d127 0610 0530  .....W.W...'...0
        0x0020:  0013 2900 bce0 1302 fa03 0300 800e 8e    ..)............
07:37:50.790109 IP (tos 0x0, ttl 255, id 989, offset 0, flags [none], proto UDP (17), length 47)
    10.1.1.140.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 19
        0x0000:  4500 002f 03dd 0000 ff11 b547 0a01 018c  E../.......G....
        0x0010:  e000 170c 0e57 0e57 001b d332 0610 0530  .....W.W...2...0
        0x0020:  0013 2900 bce0 1302 fa04 0300 8002 8c    ..)............
07:37:50.801504 IP (tos 0x0, ttl 255, id 1676, offset 0, flags [none], proto UDP (17), length 45)
    10.1.1.40.3671 > 224.0.23.12.3671: [udp sum ok] UDP, length 17
        0x0000:  4500 002d 068c 0000 ff11 b2fe 0a01 0128  E..-...........(
        0x0010:  e000 170c 0e57 0e57 0019 f157 0610 0530  .....W.W...W...0
        0x0020:  0011 2900 bcc0 1264 ab0a 0100 4000       ..)....d....@.
```

The first packet (19 byte) is received correctly.
The second packet should be 19 bytes as indicated by tcpdump, but is received as 17 bytes.

I wonder if the UDP checksum of the original 19 byte packet was wrong, but the data was not discarded? Afterwards the next packet with 17 bytes is received, and then the old data is reported with the new length?!

roberthartung, 2020-03-13:
As a temporary solution I've implemented a watchdog. While plotting and viewing the data, I came across the following interesting fact:

[image]

16:51:50
16:54:20 (+2:30m)
16:56:55 (+2:35m)
16:59:25 (+2:30m)
17:01:55 (+2:30m)

It seems like every ~150seconds the timeout occurs.

Rocketct, 2020-03-19:
Hi @roberthartung can i ask to share a short sketch version that reproduce the issue?

roberthartung, 2020-03-19:
@Rocketct I don't have another Wifi-capable Arduino here, but it looks pretty much like this:

```
#include <WiFiUdp.h>
#include <WiFiNINA.h>

const char ssid[] = "";
const char pass[] = "";

WiFiUDP udp;

void setup() {
  Serial.begin( 115200 );

  if(WiFi.status() != WL_CONNECTED) {
    Serial.print("Reason code: ");
    Serial.println(WiFi.reasonCode());      
    while (WiFi.status() != WL_CONNECTED) {
      Serial.print("Attempting to connect to ");
      Serial.println(ssid);
      // Connect to WPA/WPA2 network:
      uint8_t status = WiFi.begin(ssid, pass);
      if(status == WL_CONNECTED) {
        break;
      }
      delay(5000);
    }
    Serial.println( " Setup Multicast" );
    udp.beginMulticast(IPAddress(224,0,23,12), 3671);
  }

  IPAddress ip = WiFi.localIP();
  Serial.print("Local IP Address: ");
  Serial.println(ip);
}

void loop() {
  /// Start parsing a new packet
  int packetSize = udp.parsePacket();
  /// NOTE: this [packetSize] is sometimes *wrong* and therefore I do not read the correct amount of data afterwards
  if(packetSize > 0) {
    Serial.println( packetSize );

    /// Allocate packet data
    uint8_t buf[packetSize];
    
    /// Remaining byte to read from buffer
    int remaining = packetSize;

    /// Read data until no more bytes left
    while(remaining > 0) {
      /// Read maximum number of [remaining] bytes
      /// NOTE: it is possible that len <= remaining!
      int len = udp.read(&buf[ packetSize - remaining ], remaining );
      
      /// This is just for safety, should never happend,
      /// but otherwise we would have an overflow in the next if clause
      if(len > remaining) {
        Serial.print("received too much data: ");
        Serial.print( len );
        Serial.print(" > ");
        Serial.print(remaining);
        remaining = 0;
      }
      /// We recveived bytes
      else if(len > 0) {
        remaining -= len;
        if(remaining > 0) {
          Serial.print("  remaining: ");
          Serial.println( remaining );
        }
      } else {
        Serial.println("[WifiKNX] Unable to read packet");
        delay(100);
      }
    }

    /// This dump reveals that the contents is wrong.
    Serial.print("Packet:");
    for (int i = 0; i < packetSize; ++i)
    {
      Serial.print(" 0x");
      Serial.print(buf[i], HEX);
    }
    Serial.println("");
  }
}
```

At some point in time `int packetSize = udp.parsePacket();`  returns the wrong packet size.

Rocketct, 2020-03-19:
okk is enough to start the test

roberthartung, 2020-03-19:
Thanks a lot @Rocketct Note that this happens to be in UDP Multicast, I don't know if this causes the issue. I try to get another Arduino with Wifi so I can further test it here!

cansik, 2020-06-18:
@roberthartung I experience the same problem with an Arduino WiFi Revision 2, every now and then (5-15 seconds) I get a longer timeout.

And I also have sometimes problem when parsing OSC packages, that the package is not valid (could not yet find the prove that the size is incorrect, but I will try later). But I experience the same corrupted packages after the first error.

Maybe related: I tried a stress test and printed before each `udp.parsePacket()` to the serial port. This kind of prevented the bad messages, but after a while the Arduino did not respond to pings anymore and network requests, but it was still trying to call `udp.parsePacket()`.

Oh and I am not working with multicast.
Will keep you updated.

cansik, 2020-06-18:
Made an update to the most recent NINA FW version (`1.3.0`) still have this issue time to time (wrong package size). Interestingly, it seems that the Arduino has a problem when the UDP Client on the other side crashes / leaves and reconnects.

And more then often WiFi completely stops after sending the first few packages.

cansik, 2020-06-18:
Ok I guess I getting closer to the error now. I found out that the buffer gets overwritten by a message that is coming in later. Check out the explanation for OSC in this issue: https://github.com/hideakitai/ArduinoOSC/issues/16

The same happens with [CNMAT/OSC](https://github.com/CNMAT/OSC) so I guess it's not a library-bug but really a bug in WiFiNINA.

Is there maybe a race-condition when calling `waitResponseCmd` and already too many new data packets came in on the nina-fw side?

giulcioffi, 2020-07-02:
Hi @roberthartung, your issue looks similar to https://github.com/arduino/nina-fw/issues/48. I made a PR (https://github.com/arduino/nina-fw/pull/49) that should avoid the packet corruption.
In order to update the fw, you can use the Firmare Updater that I linked in the PR comment, following the instructions in the [README](https://github.com/arduino/FirmwareUpdater).
Let me know if this solves your problem :)

aentinger, 2020-07-09:
Solved by [arduino/nina-fw#49](https://github.com/arduino/nina-fw/pull/49).

closed 2020-07-09 by aentinger

roberthartung, 2020-07-31:
@giulcioffi I am stuck with flashing the new firmware on Windows to the Arduino Nano 33 IoT. Any complete guides on that?