arduino-libraries/WiFiNINA #113: Get Temp

rjs5327, 2020-04-17:
Does anyone have any information on the Get Temp function?

#include <utility/wifi_drv.h>
WiFiDrv:: getTemperature()

i am getting a float of 56.11 when my neighboring temp sensor is reading 75.2 F so the temp cant be in C since the chip is roughly 90C.  What i want to do is see how the low power mode effects the temp output since it is defined as a 30ma draw in low power.  So im tring to get accurate temperature measurements with the external sensor and want to compensate for the heat coming from the WiFi.  So im guessing the temperature value is interpulated rather than being in a unit.  So any information would be helpful 

TKS

milenovic, 2020-04-25:
From the nina-W102 firmware [command handler](https://github.com/arduino/nina-fw/blob/master/main/CommandHandler.cpp)

```
int getTemperature(const uint8_t command[], uint8_t response[])
{
  float temperature = (temprature_sens_read() - 32) / 1.8;
  response[2] = 1; // number of parameters
  response[3] = sizeof(temperature); // parameter 1 length
  memcpy(&response[4], &temperature, sizeof(temperature));
  return 9;
}
```
In the third line you can see that the temperature is being converted from Fahrenheit to Celsius. So your temperature should be 56.11 degree Celsius or 132.998 Fahrenheit.
Now where exactly the sensor is located I could not find. Perhaps in the WiFi module chip itself?

rjs5327, 2020-05-01:
It has to be.  I had a typo on the earlier comment that was 90F not C  . Id say this can be closed unless someone has a answer to where the temp measurement is physically.  My guess is you are right it is in the chip and the sensor is so small it is seeing a trace temp rather than the average chip temp.

closed 2020-05-01 by rjs5327