arduino-libraries/WiFiNINA #114: Is FTP over TLS be possible?

milenovic, 2020-04-24:
Hi everyone,

After seeing that ConnectSSL() exists and that it can be used for MQTT among other things I was wondering if FTP over TLS can be used as well.
My naive attempt at just exchanging WiFiClient with WiFiSSLClient fails at the very first step which is client.connect(server, port) despite of trying both Implicit and Explicit FTP configuration on the FTP server.

My experience with SSL and TLS is very limited, so bear that in mind, but I don't see a major problem in having TLS when SSL works. Am I correct? Is this even possible?

Best,
Milan

milenovic, 2020-04-24:
I made a bit of progress but stuck again. At least now I have more specific question :)

If I use explicit TLS, then FTP server expects us to connect first without TLS, which I sucessfully do with Connect() and then I send a command AUTH TLS after which we get a response
> 234 AUTH command ok. Expecting TLS Negotiation.
And here I am stuck. How can the client now initiate TLS handshake and transition to TLS withour re-connecting? Is this possible?

If I use implicit TLS, then I am able to ConnectSSL() and I can send FTP commands and receive correct responses until I reach the point when data connection should be opened.
That is, when I send the command PASV, I receinve the reply correctly:
> 227 Entering Passive Mode (195,144,107,198,4,8).
Which informs us how to open the data connection (IP 195:144:107,:98 and port 4*256+8=1032).
Now I use another client to open the data connection, and that does not work with ConnectSSL(). It does connect with Connect(), but then of course the TLS handshake never happens, and the conection times out because we never performed the handshake and issued any commands.
Here I am not entirely sure what the problem is, but I suppose that server again wants us to connect, and then later proceed with TLS???

Best,
Milan

milenovic, 2020-04-29:
So... motivated by no quick reply here, I decided to take time to "properly" understand this and I made it!
What was missing was a way to perform TLS handshake independently of opening a TCP socket because for explicit FTP over TLS to work we need to send AUTH TLS over unencrypted channel, get the response, and then perform a handshake.
At the moment this is not possible down on Nina firmware level and by extension, not possible in this library. So I modified both, and got it working, although almost definitely not in an optimal way.
If someone would have time and patience to review my code it is available in my forks of [nina-fw](https://github.com/milenovic/nina-fw) and [WiFiNINA](https://github.com/milenovic/WiFiNINA).

In case of implicit FTP over TLS, WiFiSSLClient actually works. I had a bug in my code that sends actual FTP commands.

PS: My FTP client example that utilizes added functionality needs some final polishing, so I will post that in the coming days as my time allows.

maha-haji, 2020-11-05:
Hi - I am trying to do the same thing. Would you be able to share your code? Thanks!

maha-haji, 2020-11-05:
Hi - I think I am trying to do something very similar. I need to send a CVS file to an FTP server that requires connection via FTPS (FTP with TLS AUTH). Do you have any tips of how to make this happen? Or could you share your code? As you mentioned WiFiSSLClient alone does not seem to do the trick. Thanks!