arduino-libraries/WiFiNINA #117: When called frequently, UDP parsePacket() causes access point failure

BrianG4, 2020-05-20:
Setup
-
1 x MKR WiFi 1010 as WiFi access point (hub),
1 x MKR WiFi 1010 connected to ap (node).

Both Arduinos using firmware version 1.3.0 and programmed by Arduino IDE 1.8.12 on Linux.

Node sends UDP message (counter) to hub, hub receives message, updates some counters and flashes builtin LED over and over. Node sends messages as often as possible.

Timing
-
Without any restrictions, the hub's UDP server quickly stops working (< 1 minute) and causes the network to disconnect clients.

Started looking at delays because of nina-fw issue 18: https://github.com/arduino/nina-fw/issues/18 (closed in FW 1.2.4)

Adding delays to reduce frequency of checks to the UDP parsePacket() as well as adding a delay between parsePacket() and read() can improve stability. (10ms UDP check interval, 1ms delay before read() - works better, but I have not tested it exhaustively.)  I am not sure what is causing the problem and therefore not sure if this is a reasonable fix.

Sketch
-
I used the following sketch to investigate / define the problem.
The first device turned on starts the AP as hub, following device joins the network as node.

```
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>

#include "arduino_secrets.h"

#define UDP_CHECK_INTERVAL 0 // check udp interval (ms)
#define UDP_DELAY_AFTER 0 // after parsePacket(), before read()

//works better with UDP_CHECK_INTERVAL = 10 && UDP_DELAY_AFTER = 1

#define UDP_SEND_PERIOD 0 // 0 = sending as fast as possible

// network
char ssid[] = SECRET_SSID;
char pass[] = SECRET_PASS;
IPAddress apIP(192, 168, 4, 1); // not explicitly set, relies on defaults
int status = WL_IDLE_STATUS;

// UDP variables (UDP:2048)
int UDP_port = 2048;
WiFiUDP Udp;
char packetBuffer[255 + 1];
char udpMsg[] = "message:";

// flasher
int led =  LED_BUILTIN;
bool led_on = false;

// timers and counters
unsigned long timer_next_UDP_msg_send;
unsigned long timer_next_UDP_chk;
unsigned int counter = 0; // message being sent from 'node'
unsigned int missed_count = 0;
unsigned int udp_msg_received_count = 0;
unsigned int udp_sending_error = 0;
unsigned long millis_started = 0; // time program started - millis() at end of setup()

enum e_wifi_mode{
    wm_hub,
    wm_node
} wifi_mode;

void setup()
{
    // leave time to clear output on Serial monitor
    delay(5000); 
[... 319 lines cut ...]
    
    Serial.print("counter: ");
    Serial.print(counter);
    
    if(wifi_mode == wm_hub)
    {
        Serial.print(", missed_count: ");
        Serial.print(missed_count);
        Serial.print(" (");
        Serial.print(missed_count / (counter / 100));
        Serial.print("%)");
        Serial.print(", received: ");
        Serial.print(udp_msg_received_count);
    }
    else
    {
        Serial.print(", sending errors: ");
        Serial.print(udp_sending_error);
    }
    Serial.print(", WiFi.status(): ");
    Serial.println(WiFi.status());
    Serial.println();
}
```
hope this helps.

OlliSchm, 2020-06-03:
Hey Brian, same problem here.

I have the same setup consisting of two MKR WiFi 1010. One in Ap-Mode and the other connected to it, talking over UDP Port 2390.
Both on 1.3.0. The AP also hosts a webserver.

After some time the AP crashes and has to be started again by calling `WiFi.end()` and `WiFi.beginAP()`.
With `UDP.begin(Port)` it's possible to restart the UDP-Connection. But it's not possible to restart the webServer bei calling `webServer.begin()`.

When starting another UDP-Connection or a second webServer, it takes much less time to crash.
(Btw. Arduino says the connection to the module is lost)

Excuse formal errors I am completely new here.

Best regards, Oliver

BrianG4, 2020-06-04:
Hi Oliver,

Thanks for the message, I am new here too.

It seems there are several, possibly related issues with the wifi on the MKR1010, I have tried to isolate one example. As you pointed out, restarting the udp server does seem to extend operation, but does not overcome the issue.

The approach I am taking is the idea that there is a background task that needs an unusually long time to finish what it is doing. Therefore spacing out checks and adding delays reduces the stress on this task. I don't know if this just kicks the problem further away in time, or if it lets the task complete and achieves stability - I keep interrupting my tests.

I have encountered the problem you described with a tcp http server crashing too. I noticed that sending http messages using client.write() was more stable than using client.print() and that sending smaller strings was better (40 chars at a time). Adding some delays also seemed to help. The web pages take longer to load, but more pages can be sent.

example tcp send for char array (you could use substr if using a string)
`char string_to_send[SIZE];`
```
#if defined(ARDUINO_SAMD_MKRWIFI1010)
    int str_len = strlen(pStr);
    char* str_ptr = (char*) pStr;

    int chars_to_send = 40; // seems better limiting to 40 chars at once - still testing

    chars_to_send = (chars_to_send > str_len) ? str_len : chars_to_send;
    //only send as many as needed

    while (str_len > 0 && str_len >= chars_to_send)
    {
        client.write(str_ptr, chars_to_send);
        
        delay(1); // give time to do things...
        
        str_ptr += chars_to_send;
        str_len = strlen(str_ptr);
        chars_to_send = (chars_to_send > str_len) ? str_len : chars_to_send;
    }
#else
    //method using client.print() for other hardware
#endif
```
Best regards, Brian

BrianG4, 2020-07-09:
Solved by [arduino/nina-fw#49](https://github.com/arduino/nina-fw/pull/49)

closed 2020-07-09 by BrianG4