arduino-libraries/WiFiNINA #118: WiFi Connection to Mesh

DrUdoZucker, 2020-05-24:
I believe there might be a problem using Nano 33 ioT in a Mesh environment. Let me explain: I have a Mesh consisting of 1 Master and 4 Mesh devices, all connected to each other by wires and acting as Access Points using an idential ssid.  The Mesh works at 2.4 & 5GHz.

Inside the mesh I use 4 Nano 33 IoT close to these Mesh access points, which measure sensors and are connected to by a server every minute. The problem is, that these Nano 33 IoT either connect to the Master at very low speed or connect not at all, and if they are at the limit of the master's range, they disconnect all time. 

It is, as if Nano 33 IoT does not "see" these Mesh Access Points. I have scanned the Arduino Forum and have seen several note about Nano 33  IoT disconnecting from time to time, and some mention the same hardware, I use: FritzBox 7490, 1200 & 2400.  FritzBox is a major supplier mesh devices in Germany and I can see that all Mesh Access Points do work, because PCs, tablets and mobile phones connect to them in a stable and repetitive manner.

RobbyDeLaet, 2020-06-27:
I have more or less the same issue with the Arduino MKR 1010 WiFi and an Amplify mesh.
Sometimes I get a connection ,although not with the strongest access point. It seems to have a preference for a specific access point. Only when I pause this access point another one is choosen. Then it runs for a couple of hours, only to get stuck reconnecting once more.

Technical stuff:
Arduino MKR 1010 Wifi
WiFiNINA version 1.5
The process
 - connect to WiFi
 - read out a sensor
 - send data to webapi
 - disconnect wifi
 - go to sleep
 - wake up after 30 seconds
 - loop

capndirk, 2022-01-29:
I had the same problem. My solution was to open an additional network on just one AP (using power LAN and one Wifi AP) and that solved the stability and connectivity problems. I'm using an Arduino Uno WiFi Rev. 2

wasteofpaint, 2025-04-17:
This behavior could be easily fixed with a quick change to the NINA Firmware (Not the library). I'm hopeful, (but not optimistic at this point) that it will be fixed in the upcoming NINA FW release.

The firmware should be configured to use WIFI_ALL_CHANNEL_SCAN, but instead it's set to WIFI_FAST_SCAN. 
fixing this is just a one line change in this file NINA FW: arduino/libraries/wifi/src/WiFi.cpp

See below for more info:
https://github.com/arduino/nina-fw/issues/75

closed 2025-04-17 by per1234