arduino-libraries/WiFiNINA #125: WiFiStorage class file.write() error on MKRWiFi1010

AppsByDavideV, 2020-08-06:
Hello,
I am trying to use the Wifistorage class on my MKRwifi1010 to save some settings.
I started with the example coming with the last library version.
I am experiencing errors writing files to the flash: most of the time the last 1 or 2 bytes of the string ( or char array, I tried it also) are written as '0'.

Find attached the modified example with the output from the serial monitor.

During the tests I found a workaround, maybe it can help you in the problem finding: if you write one byte at a time they are written correctly (In my case it works)

Library : Wifinina, version: 1.6.0, chip: MKRWiFi1010, IDE 1.8.13

[WiFiStorage.zip](https://github.com/arduino-libraries/WiFiNINA/files/5033353/WiFiStorage.zip)

ibslpaul, 2021-02-19:
I note your issue is closed. I have similar problem. See Arduino Forum> Using Arduino> Storage> WiFiStorage example from WiFiNiNA - Problem found with Example code.
Was the underlying problem fixed or do you rely on your work around? I have found that using different filenames and differing filename lengths vary the number of 'lost' characters read back from file.

AppsByDavideV, 2021-02-20:
Hi,
the issue has been closed without any answer.
I didn't try the very last version of the library, maybe it has been solved, but no mention about it in the changelog.

I'm still living with my workaround, for now.

per1234, 2021-02-20:
Hi @AppsByDavideV and @ibslpaul. This issue is still open. The other issue you opened in the WiFiNINA_Generic library is the one that was closed: https://github.com/khoih-prog/WiFiNINA_Generic/issues/5

Because you mentioned this issue in https://github.com/khoih-prog/WiFiNINA_Generic/issues/5, GitHub shows a "AppsByDavideV mentioned this issue" reference to it, which includes a status indicator, but that status indicator refers to the status of https://github.com/khoih-prog/WiFiNINA_Generic/issues/5, not to the status of this issue https://github.com/arduino-libraries/WiFiNINA/issues/125.

So if you dispute the closure of https://github.com/khoih-prog/WiFiNINA_Generic/issues/5, please discuss it in that issue, since that is a separate project from this one (though derived from this library of course).

AppsByDavideV, 2021-02-20:
Hi per1234,
You are right, my mistake.
It has been closed in the other project but not fixed anyway.
Because that project depends on this one, probably the author cannot fix it.

thanks for clarify that and keep up the good work!.
Davide

mikaelj, 2023-03-20:
Any updates on this?

toreniebuhr, 2023-04-18:
I am not sure if it is the same issue, but when writing to the file i added 2 to the length to make room for the null termination.

 `file.write(test.c_str(), test.length()+2);`

mikaelj, 2023-05-09:
I'm writing binary data, not a string. :)

amitabhkant, 2023-09-13:
I just hit this bug it seems, so it hasn't been resolved yet... or am I missing something here.

Library : Wifinina, version: 1.8.14, chip: MKRWiFi1010, IDE 2.2.1

amitabhkant, 2023-09-13:
@toreniebuhr Why are you adding 2 and not 1 (for null termination) ? From what I can test, adding 1 works , rather adding 2 gives out extra junk characters in some cases.