arduino-libraries/WiFiNINA #128: Find used IP addresses

EvilGarfield, 2020-09-05:
Hello
Sorry if this is dumb. I'm using wifinina with an Arduino Uno WiFi rev2 in access point mode. I've configured devices to automatically connect to the created network if it's available.
Now on the Arduino side I'd like to regularly see how many devices are connected and what IP addresses have been assigned to them. How can I do that? I tried to go through the code but can't find a function returning that.

remacbh, 2022-05-04:
Same Problem here... is there a Way now?

rcds2, 2024-06-11:
It should be "IPAddress ip = WiFi.localIP(); ".
But it hangs on my nano connect rp2040 when it is called on Pi Pico SDK (philhower). Works good when using MBED OS.