arduino-libraries/WiFiNINA #134: Problem with WiFiSSLClient

salexander2, 2020-10-09:
Hi, i have problems with connecting to a specific webside.
Others work but the website "www.aruba.it" and others sites hosted by aruba.it doesn't.

I've updated the SSL root certificates from google.com, aruba.it and others
Is there any way how i can debug this? Error Messages or Debug Messages to help me what I'm doing wrong?

Thank you very much for your help and time!

Arduino UNO WiFi Rev2
NINA firmware version 1.4.1
NINA library version 1.7.1

CroceVerdeCasale, 2023-08-16:
Hi! I have same problem right now.
@salexander2: did you sort out the issue? If yes, how?

Thank you in advance!

Tommaso

MSchmidt, 2025-04-03:
Pretty old but probably still an issue: it helps to upgrade to the latest Arduino firmware for your board. In case this is not enough you can add root certificates. It's mist likely because of outdated ciphers that some pages work and some do not.