arduino-libraries/WiFiNINA #138: WiFiNINA with VL53L0X

kovapatrik, 2020-11-09:
### WiFiNINA with VL53L0X

I tried to use the VL53L0X lidar sensor with a WiFi AP webserver, and the wifi status randomly changed to 255 then 0. If I commented out the lidar's code, the wifi worked without any problem.

### Details
Library Version: 1.7.1
Firmware Version: 1.4.1

CptHolzschnauz, 2022-05-29:
I use the Sparkun lib for the lidar without this issue

samuelkl, 2024-04-18:
I am also having this issue with a different sensor and the [OPT3101 library](https://github.com/pololu/opt3101-arduino). Did you manage to fix it?

kovapatrik, 2024-04-18:
I think in the end it turned out that I kind of fried my Arduino as it did weird things other than this. At one point it might got more current than it should have had.