arduino-libraries/WiFiNINA #140: Arduino Nano 33 IoT firmware version not available

tikonen, 2020-11-20:
Hi,

WiFiNINA example sketches complain I should update my firmware. They expect to see version 1.4.0.

Output from WiFiNINA->Tools->CheckFirmwareVersion
`Firmware version installed: 1.3.0
Latest firmware version available : 1.4.2

Check result: NOT PASSED
 - The firmware version on the module do not match the
   version required by the library, you may experience
   issues or failures.`

But this firmware is not available in Arduino IDE 1.8.13. Latest is 1.3.0. What gives?
[image]

facchinm, 2020-11-20:
Hi @tikonen ,
latest firmware are available in IDE nightly or, as a standalone package, in https://github.com/arduino/FirmwareUpdater .

tikonen, 2020-11-20:
Thank you! Please consider adding this information also in the WiFiNINA upgrade page. https://www.arduino.cc/en/Tutorial/WiFiNINAFirmwareUpdater

closed 2020-11-20 by tikonen

ocrdu, 2020-11-21:
This is an ungoing story that needs fixing, but in this specific case, 1.4.2 probably isn't there yet; the last version of WiFiNINA wants it, but was released before 1.4.2 was available, see https://github.com/arduino-libraries/WiFiNINA/commit/edf9cd77dad182a0094916a0c76517d6bdb43bec

Just remove the version check from the code and see if it works.

tikonen, 2020-11-22:
At least examples I tried (WiFi scan and web server) did work without updating the firmare. For some reason I can't connect to one of the wifi networks here but don't know if it has anything to do with the firmware.

orlaude, 2020-12-15:
I can't find the Firmware 1.4.2 for my Arduino Uno WiFi Rev. 2 in the Firmware Updater Tool on Arduino IDE 1.8.13. I use the WIFININA Library 1.8.0. 
In the download file on [https://github.com/arduino/FirmwareUpdater](https://github.com/arduino/FirmwareUpdater) is the new Firmware  1.4.2.
Can anyone tell me how to use the new firmware the firmware updater ?

Gostas, 2020-12-20:
@orlaude Firmware version 1.4.2 is not available yet, I believe it is releasing soon as @ocrdu said.

ghrom, 2021-01-09:
Firmware version 1.4.2 still not available, I hope all is well with the maintainer?