arduino-libraries/WiFiNINA #141: Can't use RGB Led when using BLE on MKR1010

MCXIV, 2020-11-26:
Hi everyone,

**Tested on Arduino MKR 1010**

Following the issue #24 , I tried to control the RGB Led, while connected to a BLE device.

Using a bit of _Example/ArduinoBLE/Peripheral/ButtonLED_ 

Unfortunately, it didn't work. Here is a piece of code : 

`
void setup() {
  Serial.begin(9600);
  WiFiDrv::pinMode(25, OUTPUT);  //GREEN
  WiFiDrv::pinMode(26, OUTPUT);  //RED
  WiFiDrv::pinMode(27, OUTPUT);  //BLUE

  pinMode(ledPin, OUTPUT); // use the LED as an output
  // begin initialization
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");

    while (1);
  }
  
  // set the local name peripheral advertises
  BLE.setLocalName("ButtonLED");
  // set the UUID for the service this peripheral advertises:
  BLE.setAdvertisedService(ledService);

  // add the characteristics to the service
  ledService.addCharacteristic(wLedCharacteristic);
  ledService.addCharacteristic(slideCharacteristic);

  // add the service
  BLE.addService(ledService);

  wLedCharacteristic.writeValue('0');
  slideCharacteristic.writeValue('0');

  // start advertising
  BLE.advertise();

  Serial.println("Bluetooth device active, waiting for connections...");
}
`

If we use 
`
WiFiDrv::digitalWrite(25, HIGH);
`
before
`
BLE.begin()
`
it works perfectly, but if we use it after, it doesn't work and it's blocking our program.

MCXIV, 2020-11-27:
**Hi,**

After going a little bit further into that problem, I though that *rebooting* the BLE service would do the job.

It doesn't. Adding : 

```
BLE.stopScan();
BLE.end();
for(int y = 1; y <= 254; y++){
      WiFiDrv::analogWrite(26, y);
      delay(1);
 }

for(int y = 254; y >= 0; y--){
      WiFiDrv::analogWrite(26, y);
      delay(1);
}
BLE.begin();
BLE.scan(0);
Serial.println("it works");
```

Well, it never reaches "it works", and the RGB Led doesn't light up.

**Maybe it's link with the fact that the MKR1010 can't use BLE and WIFI simultaneously ?**

MCXIV, 2020-11-27:
#https://github.com/arduino-libraries/ArduinoBLE/issues/3#issuecomment-445912224

That explains.

andewx, 2023-11-27:
Thanks I was having this same error and never able to even see that it was the BLE causing the issue. Documentation needs to state this!