arduino-libraries/WiFiNINA #144: client.connect() waits 28,5 seconds if cant connect

Magerquark, 2020-12-08:
Version: 1.4.2
Devide: Arduino Nano 33 IOT

Hello folks, if I use client.connect to connect to a specific device in my Network its takes exactly 28500ms till it returns false when the device is not connected to the network. While this time the program is stuck cause it waits for the method to finish.

For me those 28,5 Seconds are too long I want it to be like 3-4 Seconds so I can quickly give an output that the Device coud not be found in the network and thats it.

The problem is I dont really understand the Code of the connect method:

The comment there says it should take like 4 Seconds for the connection to close?
The code says to wait 10 seconds??
My Serial says its exactly 28,5 seconds???

If I just change the code in the wifinina library from 10000ms to 1000ms it reduces the time exactly for 9 seconds so it still takes 
19,5 Seconds to return false.

I have no fucking idea where those 18500ms get from ?

Does anybody has a clue how I get it to just try to connect for like 4 seconds?

sincerly
Alex

lasselukkari, 2020-12-30:
This just an assumption that I have not verified. The nina-fm running on the ESP32 chip uses the Arduino framework too. I don't see the timeout value being propagated there at all. At least the [CommandHandler.cpp](https://github.com/arduino/nina-fw/blob/master/main/CommandHandler.cpp) does not seem to have a functionality to change this.

I just bought a the MKR WiFi 1010 board recently because user of my library had some problems with it. I have been surprised to see how bad the software quality is considering this a official Arduino board that costs a lot of money. In this state I would not recommend this software/hardware stack to anyone. For many people Arduino is their first touch to embedded systems. It sad to think that so many people are having problems with this sort crap. The whole point of the simplification the Arduino provides should be the things just work. Sorry for the rant...

lasselukkari, 2020-12-30:
The ESP32 Arduino core has a version of connect with timeout: 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.h#L51-L54

The version used in nina-fm does not:
https://github.com/arduino/nina-fw/blob/master/arduino/libraries/WiFi/src/WiFiClient.h#L36-L37

So most likely the version used in nina-fm just uses the default value set in the esp-idf.

winner10920, 2021-05-06:
I think it may have to do with the fact that the Samd communicates with the nina module over SPI. There are two hidden delays of 10000 deep in the code that have to do with waiting for the module, so perhaps the nina module locks up from trying to connect and that long delay is from the module not communicating

lawtalker, 2021-12-02:
Are there any plans to address this? It would be great if there was a way to give up on a connection more quickly.

Uup115, 2022-01-10:
I've been struggling with this for quite a while.  I haven't been able to find a workaround.  See here: https://github.com/arduino-libraries/WiFiNINA/issues/207

mba1959, 2023-01-02:
I have the same problem, so hopefully something will happen...

JAndrassy, 2023-01-02:
you can make this modification to shorten the timeout to 18 seconds
https://github.com/arduino-libraries/WiFiNINA/pull/224/files