arduino-libraries/WiFiNINA #148: where to find wifi firmware 2.4.2?

FedeBev, 2021-01-09:
Hi there,

the WiFi firmware version required by v1.8.0 is 2.4.2, however the latest firmware version released by [WiFi101-FirmwareUpdater-Plugin ](https://github.com/arduino/WiFi101-FirmwareUpdater-Plugin ) is 2.4.1.

where the 2.4.2 version can be found?

Thanks

jandrassy, 2021-01-09:
you meant 1.4.1 and 1.4.2

FedeBev, 2021-01-09:
> you meant 1.4.1 and 1.4.2, not 2.4.1 and 2.4.2

Sure, sorry my bad!

aentinger, 2021-01-11:
Hi @FedeBev :wave:
Unfortunately version `1.4.2` took far longer to be released than I ever thought it would take (it depends on merging https://github.com/arduino/nina-fw/pull/57). Since it might still take a little while for this PR to be merged I'm correcting `WIFI_FIRMWARE_LATEST_VERSION` back to `1.4.1` in #149 as this version actually is available. Sorry for the confusion 😊

PR #149 (2021-01-11): Bugfix: The latest available wifi firmware version is in fact 1.4.1, not 1.4.2
This fixes #148.
closed 2021-01-11 by aentinger