arduino-libraries/WiFiNINA #159: .connected() for WiFiClient object still  false after client disconnected

TimAtFacture, 2021-03-19:
.connected() doesn't seem to return false even when a client is disconnected (powered down).

I am using a sketch which implements a WiFi server (WiFi Modbus TCP Server LED example from Arduino Modbus) on an MKR 1010. This sketch traps the program in a

    while(client.connected())

loop after a client connects. The client I am using is an MKR 1000, (connecting on a different WiFi network, using the WiFi101 library). If it's the first time trying to connect to the server since it's been reset I don't have any problem. When I power down the client while the server is still running, the server's code never exits the while loop, and subsequent attempts to connect to the server aren't successful.

When I reverse the rolls (MKR 1000 as the server, using the WiFi101 library, MKR 1010 as the client using the WiFiNINA library) client.connected() returns false a few seconds after I power down the client, as I expected.

For my own purposes, I have fixed this with a simple keep-alive timer based on client.available(). Mostly just making a note of unexpected behavior.

jandrassy, 2021-03-19:
https://www.arduino.cc/en/Reference/WiFiNINAClientConnected

TimAtFacture, 2021-03-19:
Thanks for the reply jandrassy,

I tried a quick experiment, adding the following code

  static uint32_t next_print = millis();
  if(millis() >= next_print)
  {
    next_print = 1000 + millis();
    Serial.println(client.status(),DEC);
  }

Inside the while(client.connected()) loop. Output was only '4', even after powering down the client. Not sure if this is a solution.

jandrassy, 2021-03-20:
> Thanks for the reply jandrassy,
> 
> I tried a quick experiment, adding the following code
> 
> ```
> Serial.println(client.status());
> ```
> 
> Inside the while(client.connected()) loop. The output was only '4', even after powering down the client. Not sure if this is a solution.

powering down the client doesn't make the connection close. it closes then after some long timeout. normal closing of TCP connection requires some acknowledgments exchange

Jauwaad, 2021-04-20:
I am seeing this with version 1.4.3 of nina-fw but doesn't exist in 1.3.0

Jauwaad, 2021-04-21:
@TimAtFacture which version of the nina-fw are you using?