arduino-libraries/WiFiNINA #165: Arduino NANO 33 IoT - WiFi module stopped working

asiemasz, 2021-03-26:
Hi,
Im using new board for 2 days and I have some problem with Wifi/BLE module. It seems not  to work. 
BLE.begin() returns 0 and WIFI.setup() return WL_NO_MODULE code. Yesterday everything worked fine. I'm powering my Arduino just using USB cable and laptop or external charger (12 V).

Is module broken or I can fix it somehow?

per1234, 2021-03-27:
Closing because we are providing support on the forum thread. If the forum discussion reveals there is a bug in this library, I'll be sure to reopen this.

closed 2021-03-27 by per1234

goofyseeker311, 2025-07-13:
I had the same issue, but somehow re-flashing the arduino nano 33 iot firmware made the wifinina module working again.