arduino-libraries/WiFiNINA #184: error: 'PinStatus' does not name a type

yashm, 2021-06-09:
Error: WiFiNINA\src/utility/wifi_drv.h:293:12: error: 'PinStatus' does not name a type

WiFiNINA at version 1.8.10

FreeGoldRush, 2021-07-01:
Is there an update for this?  WiFiNINA_Generic compiles fine but the library just hangs when running on the PyPortal.  We currently have no solution for using Wifi on our PyPortal.

Thanks!

jhoughjr, 2021-07-07:
doesn't work with platform io in cpp either.

Darvesh7, 2021-08-13:
I tried on Nano 33 iot. It is not working. I am using platform.io. My software has been working fine for many months now it wont even compile. 

.pio\libdeps\nano_33_iot\WiFiNINA\src/utility/wifi_drv.h:293:12: error: 'PinStatus' does not name a type

Darvesh7, 2021-08-13:
Version 1.8.2 does not have this issue. Its only v1.8.3

xvs03, 2023-03-20:
Hello, I have been following the discussion. I have now installed version 1.8.12 in the Arduino IDE and I still get the compilation error. I am using an Arduino UNO WiFi Rev.2 in my project. This is an expensive original, why don't the original libraries work there.

wifi_drv.h:293:12:
Error message on Arduino UNO Wifi:
error: 'PinStatus' does not name a type
static PinStatus digitalRead(uint8_t pin);

On the MKR WiFi 1010 the compilation with version 1.8.12 works fine.

By the way, I have tested older versions. None of them worked.

xvs03, 2023-03-20:
I have found a solution to compile the Arduino UNO WiFi Rev.2 board.
You have to search for "Arduino megaAVR" in the board manager and install it, because the board is based on a completely different chip than the Arduino UNO WiFi (Rev.1).
[image]

It would be nice if this would be published on the manufacturer's site.

MJCoolTech, 2023-05-30:
I also have this issue in PIO using the following ini:
```ini
[env]
framework = arduino
lib_deps = arduino-libraries/ArduinoIoTCloud@^1.11.1
build_type = debug
lib_ldf_mode = deep+

[env:esp32dev]
platform = espressif32
board = nodemcu-32s
```

Now reverting to v1.8.3.

TheOdOgre, 2025-04-13:
I'm having the same problem at the moment, WifiNINA v1.9.1

JAndrassy, 2025-04-14:
> I'm having the same problem at the moment, WifiNINA v1.9.1

which Arduino board?

TheOdOgre, 2025-04-14:
> > I'm having the same problem at the moment, WifiNINA v1.9.1
> 
> which Arduino board?

Hi! I found out the problem, I simply had the wrong arduino board selected in the tools tab. Thank you for your reply!