arduino-libraries/WiFiNINA #192: UDP 20ms latency

ninouing, 2021-06-21:
Hi all,

I'm trying to send data as fast as possible over UDP between two MKR WiFi 1010 (I also tried with Nano 33 IoT, I get same results), using the UDP example WiFiUDPSendReceiveString as a starting point. I was hoping to reach a latency of few ms, which should be possible with UDP.

What I did was to start an Access Point in one MKR (let's call this MKR "Server"), and connect the other one over Wifi (let's call it "Client"). Then, after starting UDP on both on port 80, I started sending data from the Client to the Server periodically, according to a "timer" variable. On the Server side, I check for incoming data periodically (also here according to a "timer" variable), and once I receive one, I print the time elapsed since the previous reading.

The result is that, no matter if I send data every 1, 5, 10, 15, or 20 ms, I ALWAYS read data on the server side every ~20ms, with several packets lost in case I send data faster than 20ms. Note that all of this is independent on the timer I insert on the server side. What I mean by that is that, if on the server side I check for data every 0 (so basically every loop), 1, 5, 10 or 20 ms, I keep receiving data every ~20 ms. I even tried to change the code and simply put a delay() function on the server side instead of the timer check but nothing changes.

Few remarks:

- If I perform "ping" between the two arduinos I get 0ms, with occasionally 10ms.
- I have updated the Nina firmware to the 1.4.5 version.
- The size of the packet sent seems not the issue. I sent a full phrase or a single value, I still get the ~20ms delay

What surprises me is how reliably I always get 20ms latency, also with different MKR boards, which makes me believe this is not something related to a random issue (like packets lost along the way), but more something systematic, like some delay imposed by the WiFiNINA library. I tried to look at the various function but I couldn't find the cause of this delay. 

Can you please help me figure out how to solve this, or at least confirm me that this is the fastest possible communication I can have over UDP using Arduino?

Here you have my simplified code in case you would like to try it yourself (in this example I send data every 10ms and check server side every 5ms for updates):

Client:
``
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>

//WiFi variable setup
char ssid[] = "My_Network";     // your network SSID (name)
char pass[] = "My_Password";    // your network password (use for WPA, or use as key for WEP)
int status = WL_IDLE_STATUS;

//UDP variables setup
unsigned int port_udp = 80;      // local port to listen on
char packetBuffer[256]; //buffer to hold incoming packet
char  data_to_send[] = "h";       // a string to send back
IPAddress ip_server(192, 168, 4, 1);
WiFiUDP Udp;  

//Time variables
unsigned long previous_timer;
unsigned long timer=10000;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }
  Serial.println("Connected to WiFi");
  printWifiStatus();

  // start the UDP on port 80
  if(Udp.begin(port_udp)){
    Serial.print("Udp started on port: ");
    Serial.println(port_udp);
  }else{
     Serial.println("Udp not started");
  }

  //I added the noLowPowerMode comand to be sure we are not in lowPower
  WiFi.noLowPowerMode();

  //Inizialization time variable
  previous_timer=micros();
}

void loop() {
  
  //Send data according to timer 
  if(micros() - previous_timer >= timer){
    previous_timer = micros();
    
    //Sending data, with checks to see if sending is successiful or not
    if(Udp.beginPacket(ip_server, port_udp)){
      Udp.write(data_to_send);
      if( Udp.endPacket()){
        Serial.println("data sent");
      }else{
        Serial.println("error sending data");
      }
    }else{
      Serial.println("error inizializing data sending");
    }
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print gateway's IP address:
  IPAddress ip2 = WiFi.gatewayIP();
  Serial.print("Gateway IP Address: ");
  Serial.println(ip2);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
``

Server:
``
#include <SPI.h>
#include <WiFiNINA.h>
#include <WiFiUdp.h>

//WiFi variable setup
char ssid[] = "My_Network";     // your network SSID (name)
char pass[] = "My_Password";    // your network password (use for WPA, or use as key for WEP)
int status = WL_IDLE_STATUS;

//UDP variables setup
unsigned int port_udp = 80;      // local port to listen on
char packetBuffer[256]; //buffer to hold incoming packet
WiFiUDP Udp;  

//Time variable
unsigned long previous_timer;
unsigned long previous_loop_timer;
unsigned long timer=5000;
unsigned long diff;

void setup() {

  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
 
  // Check for the WiFi Module
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }
 
  // Print the Network Name (SSID);
  Serial.print("Creating access point named: ");
  Serial.println(ssid);

 
  //Starting access point  
  status = WiFi.beginAP(ssid, pass);
  if (status != WL_AP_LISTENING) {
    Serial.println("Creating access point failed");
    // don't continue
    while (true);
  }

  // wait 3 seconds for connection:
  delay(3000);
  
  // start the UDP on port 80
  if(Udp.begin(port_udp)){
    Serial.print("Udp started on port: ");
    Serial.println(port_udp);
  }else{
    Serial.println("Udp not started");
  }
  
  //I added the noLowPowerMode comand to be sure we are not in lowPower
  WiFi.noLowPowerMode();
  
  //Inizialization time variable
  previous_timer=micros();
  
}

void loop(){
  if(micros() - previous_loop_timer >= timer){
    previous_loop_timer = micros();
    //checking for new packets 
    int packetSize = Udp.parsePacket();
    if (packetSize) {
      // read the packet into packetBufffer
      int len = Udp.read(packetBuffer, 255);
      
      //Evaluate time elapsed since last reading
      diff=micros()-previous_timer;
      previous_timer=micros();
      
      if (len > 0) {
        packetBuffer[len] = 0;
      }
  
      //Print results
      Serial.print("Received:");
      Serial.print(packetBuffer);
      Serial.print(" Time: ");
      Serial.println(diff);
    }
  }  
}
``

Thank you in advance for your help.

Nick-Zhang1996, 2023-02-28:
I'm also running into this issue.

Nick-Zhang1996, 2025-03-27:
I would like to add that from my probe this seems to be a problem on the esp32 firmware that's in the ublox Nina chip, because if I can check Udp.parsePacket() at a high frequency and it will consistently return 0 until a 20ms delay has passed. If I have to speculate I would guess the NINA chip is on a power saving / sleep mode and wakes up every 20ms, however I haven't figured out what it is. This is done on a Arduino Nano 33 IoT.

Nick-Zhang1996, 2025-04-01:
I have identified the source of this issue. It is not the WiFiNINA library's problem, but its counterpart, nina-fw, which is the firmware loaded to the esp32 inside the ublox NINA W102 wireless IC. If you look at the file nina-fw/arduino/libraries/WiFi/src/WiFiUdp.cpp, line 168 [https://github.com/arduino/nina-fw/blob/eca4789e11a7b89e578413f59c9d15fe6b51c6f5/arduino/libraries/WiFi/src/WiFiUdp.cpp#L168](url), you'll find that if there is pending data from the buffer, the program will delay for 1 tick FreeRTOS time, which is 10ms with current defaults. If you change the default frequency to 1000Hz, the latency will significantly decrease. I'm not sure why delay is introduced there, especially since it's executed only when there's data coming in.