arduino-libraries/WiFiNINA #200: WifiNiNA cannot handle when the same SSID is broadcasted from multiple APs

borianson, 2021-12-07:
We tried multiple MKR1010 boards with multiple different WifiNiNA FW revisions but could not establish a stable connection via wifi (or at the wast majority of the cases even connect). We have a grid of ~20 pcs of well working brand new Ubiquiti UniFi APs broadcasting the same SSID at our facility.

With ESP32 it is implemented to start connection with referencing the BSSID of the desired AP, but it not has been implemented with WifiNiNa. Theoretically the implemetation of that solution would solve the issue.

per1234, 2021-12-07:
Also reported at https://github.com/arduino/nina-fw/issues/75

Russell108, 2022-11-24:
I also have this issue and I guess anyone in a business environment using Access points
Ability to connect to a specific Access point where many access points share the same SSID
Currently, in Business hotel environment where the same SSID password is used across a set of access points

The Arduino MKR1010 and the Arduino IOT 33 do not connect to a sensible access point instead connect to one with a real signal.

Ideally, there would be. A function in the library to select the best access point by signal strength for a given SSID

Russell108, 2022-11-24:
It woul eb great to know if this was being worked on. It seems no one is even assigned to it as yet

borianson, 2022-11-24:
It would be so nice if someone could resolve this issue. Unfortunately it is beyond my capabilities.

JAndrassy, 2022-11-24:
@per1234 the problem is in the firmware. 
https://arduino.stackexchange.com/questions/90303/why-does-arduino-nano-33-iot-always-choose-the-weakest-wifi-bssid/90304#90304

per1234, 2022-11-24:
Thanks @JAndrassy. It seems to me that the current `WIFI_FAST_SCAN` behavior is beneficial under the most common conditions (multiple APs are present, but all have unique SSIDs). For this reason, I think the best solution would be to make it configurable, in which case the **WiFiNINA** library code will need to be changed along with [the nina-fw code](https://github.com/arduino/nina-fw), as you pointed out:

> the firmware and the WiFiNINA library have to change to support it.

This issue will be used to track the work needed in the library.

Russell108, 2022-12-16:
I disagree most professional uses in business a share SSID over multiple access points is most common.
Maybe for a hobbyist at home what you say might be true but I really question your assumption for business users.

fast scan might save the hobbiest time but for a person on a shared SSID it makes the wifi un usable. I would say usability over saving a little time should take precedence.

How is the work going

RandyPostma, 2023-02-25:
Hello, I've been experiencing the same problem.  I have multiple access points with the same SSID and my Arduino MRK1010 doesn't select the best channel.  My observation is that it seems to prefer Channel 1, or the lowest channel available. Since the WiFi.scanNetworks() functionality provides me all the information I need to identify the best AP, I've written code that scans the network, identifies the best AP to connect to and then, if WiFiNINA would support it, connect to a specific BSSID. 

Would it be possible to update WiFiNINA to allow us to use the BSSID to identify which AP we want to connect to? 

This would really help the users that have a mobile application in an environment that has numerous access points.

J1mbo, 2023-07-14:
I've just run in to this issue.

In my environment I have six Unifi APs (mixture of UAP-ACs and U6s) all broadcasting a 2.4GHz-only WPA2 SSID for 'simple' devices. The Uno WiFi Rev.2 consistently connects (and re-connects) to the most distant access point at -89dB and then of course loses connection periodically thereafter. The most distant AP is broadcasting on channel 1, but there is a closer AP also on channel 1 also broadcasting the same SSID.

I should also note that an Arduino IOT33 seems not to be affected by this issue and has chosen the closest AP.

The Uno WiFi Rev.2 is running NINA firmware 1.4.8.

Russell108, 2023-07-14:
Good luck if you talk to arduino they will just threaten you