arduino-libraries/WiFiNINA #203: Unstable wifi connectivity

zzmike76, 2021-12-21:
I am using an Arduino Uno Wifi Rev2 with a Grove shield and Atlas Scientific board to control my aquarium (lights, co2 supplier) and to measure values (ph, conductivity and temperature). If I remove any network related code the controller runs stable for months without any maintenance required, but when adding Wifi code to send data to an Influx db hosted on my raspberry the system becomes very unstable.

Locks occur every few hours (longest uptime reached is around 30 hours, but usually it hangs after 3/4 hours), and was able to reproduce the same problem on  another board (this one is a weather station with bme280, light and noise grove sensors).

Am using the latest firmware and libraries (1.4.8 and 1.8.13)

tfonfara, 2021-12-26:
Same here with Nano 33 IoT, I'm also on latest versions.

capndirk, 2022-01-29:
I had the same problem - very slow connect to the network, unstable connection. Then I've read about the problem with multiple access points broadcasting the same SSID, which I have at home (I'm using a Wifi mesh). So I added an additional network on one AP only and now disconnects and hangs are gone.